SuperHouse/esp-open-rtos #323: sdk_os_delay_us needed for ms delay?

Zaltora, 2017-01-06:
I see than some examples and libraries use sdk_os_delay_us to create delay in ms.
I think about replace them by  vTaskDelay(x / portTICK_PERIOD_MS);

sdk_os_delay is just a busy loop ?  scheduler will not change task if used ?  

If not, it can be better to change some of delay.
For example if a library need to wait a minimum 4.5 ms, i can use vTaskDelay to wait 10 ms. The library wait a little more time but overall, more treatment is done with at least two tasks.
scheduler can stop this  4.5 delay whenever after all.

resetnow, 2017-01-06:
I wouldn't rely on vTaskDelay for delays < 5ms because its resolution is only a tick, and the tick time can be changed by user.

Zaltora, 2017-01-06:
The problem with delay <5ms, that if  tick resolution > 10 ms  ?
We can do a selection between delay system with Preprocessor.
checking configTICK_RATE_HZ .

resetnow, 2017-01-06:
Not only that — I'm not sure what happens when you call vTaskDelay right after RTOS timer interrupt. I don't have access to hardware right now but you can try it using `vTaskDelay(2)` and another delay function which would delay the task by several microseconds.

```
Timer interrupt -------*-------*-------*-------*-------*-------*-------*
vTaskDelay call           *
Time                      \____/
```