SuperHouse/esp-open-rtos #325: Examples will not connect to WiFi

BlueShrub, 2017-01-11:
I'm starting to lose my brain, here.  I just got back to development after 6 months of setting it down.  After updating to the newest rev, I'm unable to get any of my hardware to work.

I'm specifically trying to get http_get to work.  I've created the private_ssid_config.h file.  I've tried specifying SSID and PSK in the main.c file directly.

I'm able to scan for APs successfully, and I'm able to run the access_point example and connect.

Any ideas what I could be doing wrong?  I've tried several different modules (I have 6 or so) -- none of them work.

gamer-cndg, 2017-01-11:
Try using the WiFi connection status code at https://github.com/SuperHouse/esp-open-rtos/blob/master/examples/aws_iot/aws_iot.c#L229 . Maybe you're getting `STATION_CONNECT_FAIL` or `STATION_WRONG_PASSWORD` or just `STATION_IDLE`. Are there special characters in the SSID or the PSK? WPA/WPA2 keys must be at least 8 chars long. Is it a WPA or WPA2 network? Does it fail with every WiFi network or only with one specific router?

BlueShrub, 2017-01-11:
Good call on calling sdk_wifi_station_get_connect_status() -- I had forgotten about that.  I will do so when I get home again tonight.

To answer your other questions:

> Are there special characters in the SSID or the PSK? 
There are a few '!' characters in the PSK, but that has never been an issue.  The SSID is all letters.

> WPA/WPA2 keys must be at least 8 chars long.
The PSK is > 20 characters

> Does it fail with every WiFi network or only with one specific router?
It fails on every access point I've tried, including two different networks (different routers).

I'll check the wifi_connect_status code and report back.  In the meantime if anyone has any other ideas, I'd be glad to hear them.

yesco, 2017-01-11:
Are you calling?

    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&config);

BlueShrub, 2017-01-11:
>Are you calling?

>    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&config);

I'm using the example http_get, which makes these calls in user_init() here:
https://github.com/SuperHouse/esp-open-rtos/blob/master/examples/http_get/http_get.c#L126-L127

I just got home and added a call to sdk_wifi_station_get_connect_status().  It returns 0, which correlates to STATION_IDLE.  That makes even less sense.  

As I said, if I run the wifi_scan example, every access point shows up as expected.  Doesn't make any sense...  To be perfectly clear, I had been developing daily for months on this platform with this same hardware and the same network configuration.  I feel like I'm missing something here.

gamer-cndg, 2017-01-11:
Can you post your exact code in a gist please?

BlueShrub, 2017-01-11:
>Can you post your exact code in a gist please?

The code can be found [here](https://github.com/SuperHouse/esp-open-rtos/blob/master/examples/http_get/http_get.c)

The change I made was changing [line 43](https://github.com/SuperHouse/esp-open-rtos/blob/master/examples/http_get/http_get.c#L43) to:
`printf("DNS lookup failed err=%d res=%p wifi=%d\r\n", err, res, sdk_wifi_station_get_connect_status());`

gamer-cndg, 2017-01-11:
I can confirm that the code works properly with my WiFi in the latest commit. Can you try the code at https://gist.github.com/gamer-cndg/01eefd148c64447161debf831b0fe628 for more accurate debugging results? If you still only get `STATION_IDLE`, something really magical is going on inside your ESP.

BlueShrub, 2017-01-11:
Output from running your code:

▒blpp_task_hdl : 3fff1e38, prio:14, stack:512
pm_task_hdl : 3fff22c8, prio:1, stack:176
frc2_timer_task_hdl:3fff4e90, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jan 11 2017 18:46:00
phy ver: 273, pp ver: 8.3

SDK version: 0.9.9, free heap 44056
mode : sta(5c:cf:7f:1c:7b:b1)
add if0
wifi_task: Connecting to WiFi "AlmostTheSame" (pw: "redacted!butlong!similar")
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.
wifi_task: status = 0
WiFi: idle.

lujji, 2017-01-12:
Try enabling auto-connect:

sdk_wifi_set_opmode(STATION_MODE);
`sdk_wifi_station_set_auto_connect(TRUE);`
sdk_wifi_station_set_config(&config);

BlueShrub, 2017-01-12:
>Try enabling auto-connect:
sdk_wifi_set_opmode(STATION_MODE);
sdk_wifi_station_set_auto_connect(TRUE);
sdk_wifi_station_set_config(&config);

Welp, that did it!  What's ironic is I remembered this function existed, and was going to try it when I got home.  I remember it occurring to me before, but dismissing it.

What I'm assuming is I set it to FALSE in a previous binary -- apparently it sticks.  Thanks everyone for your help!!

closed 2017-01-12 by BlueShrub