SuperHouse/esp-open-rtos #326: sdk_spi_flash_write   without   sdk_spi_flash_erase_sector

skegosk, 2017-01-15:
Hi, new to esp-open-rtos, have my own Ubuntu environment working and am able to successfully flash my esp-12 and get many of the examples and functions working, very impressive.

The question I have is I would like to be able to store and modify data so that it persists between power cycles and boot-ups, same as using an EEPROM with say an Arduino.

**_QUESTION 1:_** I am assuming the easiest way to do this without the need for an external EEPROM/Chip is to use some of the FLASH? 

Based on information from the rboot readme.md and the extensive reading I have done to try and educate myself, I have selected sector 0x77000 to use for my data, which is a sector after my ROM (0x2000 plus its length 0x3C000), but before the 8Mbit boundary - 0x100000.

**_QUESTION 2:_** How do I determine the most appropriate sector to use to store my data?

I am able to initialize the sector with zero data during flash by writing a blank file of 4096 bytes to 0x77000:

_esptool.py -p /dev/ttyUSB0 --baud 115200 write_flash -fs 32m -fm qio -ff 40m \
0x0 ../../bootloader/firmware_prebuilt/rboot.bin
0x1000 ../../bootloader/firmware_prebuilt/blank_config.bin
0x77000 ../../bootloader/firmware_prebuilt/blank_config.bin
0x2000 ./firmware/Sprinkler.bin 
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0040
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (90.4 kbit/s)...
Writing 4096 @ 0x1000... 4096 (100 %)
Wrote 4096 bytes at 0x1000 in 0.4 seconds (90.3 kbit/s)...
Writing 249856 @ 0x2000... 249856 (100 %)
Wrote 249856 bytes at 0x2000 in 21.6 seconds (92.5 kbit/s)...
**Writing 4096 @ 0x77000... 4096 (100 %)
Wrote 4096 bytes at 0x77000 in 0.4 seconds (90.4 kbit/s)...**
Leaving..._

In my `user_init` function I then look to see if the data is zeroed, telling me this is the first boot after flashing, I then run an initialization function to setup the defaults and attempt to write the data to FLASH.

Here is where I am missing something... I have the `sdk_spi_flash_write` and `sdk_spi_flash_read` functions working just fine, allowing me to read my data from FLASH sector 0x77000 and also write it back, however I am only able to write my data to FLASH if I first call `sdk_spi_flash_erase_sector` after each boot.

The fact that `sdk_spi_flash_erase_sector` must first be called each time the unit boots means my data is lost and defeats the point of using it.

**_QUESTION 3:_** Is it possible to write updates to my data to FLASH without first erasing the sector?

Interestingly, if I do not call `sdk_spi_flash_erase_sector` before I attempt to write my data to FLASH, the `sdk_spi_flash_write` function appears to store in flash a bitwise AND of the current data at the address in flash and the data I am attempting to store.

I see that in calling `sdk_spi_flash_erase_sector`, a call is made to `SPI_write_enable` prior to `SPI_sector_erase`. I have tried directly calling `SPI_write_enable` before `sdk_spi_flash_write` but I still get the same result which is the function storing the bitwise AND of the current data in flash at the address and the data I am attempting to store.

Thanks in advance for any assistance.

resetnow, 2017-01-15:
>QUESTION 3

No, that's how flash memory works. Erasing is different on a physical level — it requires a rather high voltage and affects a number of cells grouped in blocks. All these cells are reset to logical 1 after erasing. You can only set bits to 0 when writing.

skegosk, 2017-01-15:
Thanks **resetnow**, your response explains the behavior I was seeing, that just using `sdk_spi_flash_write` performed a bitwise AND of the current data in flash at the address with the data I am attempting to store, as the write function is unable to set logic level 1, but rather can only set logic level 0. Appreciate the quick response.

closed 2017-03-06 by UncleRus