SuperHouse/esp-open-rtos #327: sdk_wifi_promiscuous_enable

mehdilauters, 2017-01-18:
Hi, 
I already done one project using the promiscuous mode of the esp based on the official espressif sdk, but I cannot make it working using the rtos sdk. I cannot find any information on the wiki nor web.
The result is that the ````promisc_cb```` is never called
````
#include "wpa_handshake.h"
#include <stdint.h>
#include "FreeRTOS.h"
#include "task.h"
#include <string.h>
#include "espressif/esp_common.h"

void channelHop()
{
  // 1 - 13 channel hopping
  uint8_t new_channel = sdk_wifi_get_channel() % 12 + 1;
  sdk_wifi_set_channel(new_channel);
}

void frc1_interrupt_handler(void)
{
  channelHop();
}

void promisc_cb(uint8_t *buf, uint16_t len) {
}

void enable_monitor() {
  timer_set_frequency(FRC1, 10);
  timer_set_interrupts(FRC1, true);
  timer_set_run(FRC1, true);
  sdk_wifi_station_set_auto_connect(0);
  
  sdk_wifi_station_disconnect();
  sdk_wifi_set_opmode(STATION_MODE);
  sdk_wifi_promiscuous_enable(0);
  sdk_wifi_set_promiscuous_rx_cb(promisc_cb);
  sdk_wifi_promiscuous_enable(1);
}

void wpa_handshake_reset() {
  enable_monitor();
}

void scanmap_task(void *pvParameters) {
 // wait a little bit in order to be sure the init is done ( no system_init_done_cb within rtos sdk ) 
  vTaskDelay(1000 / portTICK_PERIOD_MS );
  wpa_handshake_reset();
  while(true) {
    vTaskDelay(1000 / portTICK_PERIOD_MS );
  }
}

void wpa_handshake_init() {
  
  timer_set_interrupts(FRC1, false);
  timer_set_run(FRC1, false);
  _xt_isr_attach(INUM_TIMER_FRC1, frc1_interrupt_handler);
  
  xTaskCreate(scanmap_task, (const char *)"scanmap_task", 512, NULL, 3, NULL);//1024,866
  
}
````

Thank you for any information

Mehdi

mehdilauters, 2017-02-01:
After several tests the problem was from another part of code.
Sorry for the issue

mehdilauters, 2017-02-01:
Actually it was working with the non os sdk, but still not with the rtos sdk

andrewclink, 2017-03-21:
This is because `sdk_wifi_set_promiscuous_rx_cb` does not work. I think it's in a blob somewhere. 

As a work-around, you can put
`extern sdk_wifi_promiscuous_cb_t sdk_promiscuous_cb` 
in your code and then set `sdk_promiscuous_cb` manually.

nickw444, 2018-01-07:
Been mucking around with this for the last day or so, but unable to get promiscuous mode to work - My ESP appears to do one of two things randomly as soon as I make a call to `sdk_wifi_promiscuous_enable(1)`. It either:

1. Performs a WDT reset (I imagine due to a hang)
2. Hangs and does not reset

Interestingly enough, running similar code in Arduino/PlatformIO environment does not exhibit this problem. I was really looking forward to using `esp-open-rtos` for this project.

pxdelta, 2018-01-22:
Exactly as @andrewclink quoted.
`extern sdk_wifi_promiscuous_cb_t sdk_promiscuous_cb;
sdk_promiscuous_cb=my_promiscuous_rx_callback;`
For me the callback worked.

nickw444, 2018-01-23:
Yep would love to know if there's a solution for this. I'm sad I had to go back to Arduino/PIO for my project.

ourairquality, 2018-01-23:
Unfortunately these paths appear to call back into the C library, such as `malloc()` which is not safe in that context, so promiscuous mode is not expected to work without some significant work-arounds. The main code path goes to some length to store packets in a safe pool in the NMI contect and then later safely moves them to another pool, and that might be necessary.