SuperHouse/esp-open-rtos #329: {Error} 'Connection dropped, request restart' when subscribed to AWS thing/shadow/update/delta

tbinswirls, 2017-01-26:
Hi!

I am relatively new to esp-open-rtos, and many many thanks for this great repository. I find esp-open-rtos extremely helpful in a project where I am trying to interact with a 'Thing Shadow' on AWS.

I have gotten far enough to be able to publish messages to the shadow. However, there seems to be an issue when ESP is subscribed to the 'thing/shadow/update/delta' topic. The error happens when AWS detects a difference between 'desired' and 'reported' states of the thing shadow and it sends a message to the '/delta' topic. The error message is:
'Connection dropped, request restart'

I am sure that the message being sent to the '/delta' topic reaches ESP because this error happens only when there is a difference between 'desired' and 'reported' states.

I have been successful in subscribing and receiving messages from 'non-thing' topics, but got stuck with receiving the messages from any of the 'thing shadow topics'. 

Could it be the size of the payload? The messages received by ESP from 'non-thing' topics were just 'on' and 'off'.

Did anyone face this issue?

tbinswirls, 2017-01-28:
Nevermind, did some digging around on the previous issues (#158). Realized this problem was addressed. I was able to get the data from shadow/delta topic by increasing the read buffer size in the 'mqtt_client_new' function call in aws_iot.c.

Again thanks for the great work this community is doing.

closed 2017-01-28 by tbinswirls

rodrigopex, 2017-01-30:
You could use rules to forward messages (to other topic, created by you)
from delta topic with only the information you need (desired section). I am
doing that and it is being working properly. No extra memory is necessary.
There are a lot of unnecessary information at the delta topic.

Best regards,
Rodrigo Peixoto.

only1chi, 2017-02-22:
@tbinswirls I would like to know how you were able to start publishing on things/topic with aws_iot.
I've been working on it for a couple of days now. I think I have my policies on aws set up correctly. I use openssl to test connection to the endpoint (and this is successful).
I try to use mosquitto_pub to test publishing to my aws topics, but its not working. I'm actually getting "Error: Connection is lost".
Did you run into this problem?

tbinswirls, 2017-02-22:
@rodrigopex Thanks for the clever idea...I got it mine to work by increasing the memory only by a bit. But I will keep your trick in mind for sure.

@only1chi I did not get the "Error: Connection is lost" message. I suggest first trying to publish to the topics by setting the MQTT_PUB/SUB_TOPIC as below.
#define MQTT_PUB_TOPIC "<topic name>/status"
#define MQTT_SUB_TOPIC "<topic name>/control"

For me it worked right away. Also could you post the messages ESP sends out on its serial port?

only1chi, 2017-02-22:
@tbinswirls thanks for responding.
Here is the output from my console:
`
mqtt_task: started
mqtt_task: connecting to MQTT server A1VJS5VSZIZ8WS.iot.us-east-1.amazonaws.com ... Schedule  to publish `
`Publish queue overflow `
`done`
`Send MQTT connect ... mqtt_ssl_read: TLS read error (-30848), resetting `
`mqtt_ssl_read: TLS read error (0), resetting`
error: -1 `

So it tries to connect. Its unable to connect because it has a TLS error. I did generate an EC key using the README guidelines provided in aws_iot example.

only1chi, 2017-02-22:
The error code is:
`
#define MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY                 -0x7880  /**< The peer notified us that the connection is going to be closed. */
`

tbinswirls, 2017-07-23:
@only1chi Sorry for the delayed response. Not ssure what is causing that error. Have you attached the certificate to the policy in AWS?