SuperHouse/esp-open-rtos #333: Changing hostname

lujji, 2017-02-02:
Is there any way of changing the hostname? Original SDK provided a function wifi_station_set_hostname(), which is missing in esp-open-rtos. The only workaround I found was to enable `LWIP_NETIF_HOSTNAME` and change the hardcoded hostname inside `ethernetif_init()` in lwip/esp_interface.c

kezl, 2017-02-02:
I think something in the same vein as you have done may be the only way, when I thought about trying this the most likely lead I found was a post that predates sdk_wifi_station_set_hostname being added to the official sdk:
http://www.esp8266.com/viewtopic.php?f=6&t=617
There they alter the dhcp connection code of lwip using the softap_config struct, which I think under esp-open-rtos is sdk_softap_config in include/espressif/esp_softap.h

Not tested at all ;-)

lujji, 2017-02-04:
It turns out this can be easily done without modifying any internal libraries:
`#include <lwip/api.h>`
...
`sdk_wifi_station_disconnect();`
`netif_set_hostname(netif_default, "new_hostname");`
`sdk_wifi_station_connect();`
As I mentioned, `LWIP_NETIF_HOSTNAME` needs to be enabled.

closed 2017-02-04 by lujji

skcks, 2018-01-23:
LWIP_NETIF_HOSTNAME is enable default ? wireshark capture the dhcp request always whitout hostname.  I try to fix this problem, could you give me a little help , please?

ourairquality, 2018-01-23:
There were too many reports of this breaking for people, so it was disabled in the source code. To enable this function uncomment the following line in esp_interface.c function `ethernetif_init()`: `sysparam_get_string("hostname", &hostname);` and then the hostname can be set via the sysparam db.

skcks, 2018-01-24:
thanks @ourairquality , I will try.  how to pull your last version lwip to esp-open-rtos,  make occur a error: `fatal error: lwip/debug.h: No such file or directory` .

ourairquality, 2018-01-24:
@skcks Yes there has been a bit of upstream work and a sync is needed, see https://github.com/SuperHouse/esp-open-rtos/pull/554 It could use testing and feedback.

Doodles2000, 2023-09-13:
Apologies for re-opening an old topic.
Recently upgraded my router and the devices with open rtos really struggle to obtain an IP.  On the other hand esp devices loaded with tasmota continue to work without missing a beat.
Obviously the DHCP stacks are very different, from a user point of view, the main diff I can see is that the tasmota devices are sending a hostname to the DHCP server, where as the op rtos ones don;t.

Considered setting a static hostname in esp_interface.c, but that would open up another can of worms....

To ensure unique host names are sent, would like to have open rtos devices send a prefix followed by MAC address (something like 'ESP-aabbccddeeff")

using 'esp_base_mac_addr_get()' fails, 

What would be the correct way to populate hostname with MAC address be?