SuperHouse/esp-open-rtos #335: Problem GET'ing binary file

ole-johan, 2017-02-07:
I am sending a GET message to download a binary file using the mbedtls HTTPS setup. I am successfully using POST, and GET works for small text files. 

When I try to GET a binary file that is 520kb, I get the following response when calling `mbedtls_ssl_read(&ssl, buffer, 300);` in a loop:

```
HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
X-Frame-Options: SAMEORIGIN
X-XSS-Protection: 1; mode=block
X-Content-Type-Options: nosniff
X-Download-Options: noopen
Accept-Ranges: bytes
Date: Tue, 07 Feb 2017 20:49:39 GMT
Cache-Control: public, max-age=0
Last-Modified: Tue, 31 Jan 2017 20:39:18 GMT
ETag: W/"811b8-977420404"
Content-Type: application/octet-stream
Content-Length: 528824
set-cookie: connect.sid=...; Path=/; HttpOnly
Connection: keep-alive

```
However, after receiving this header, I get the following error on the subsequent call to `mbedtls_ssl_read` :

```
ssl_tls.c:3363: bad message length
ssl_tls.c:6185: mbedtls_ssl_read_record() returned -29184 (-0x7200)
ERROR: SSL read failed. -29184 - SSL - An invalid SSL record was received

```
Any tips on how to make this work?

Thanks, OJ.

projectgus, 2017-02-07:
It'd be best to confirm with a packet capture or by enabling mbedTLS debugging, but the problem is probably that the server is sending a TLS message with more than 4096 bytes of data in it:
https://github.com/SuperHouse/esp-open-rtos/blob/master/extras/mbedtls/include/mbedtls/config.h#L2403

TLS spec says that messages can be up to 16KB, but this requires 32KB of RAM on the ESP8266 (16KB RX buffer + 16KB TX buffer.) So by default it's turned down to 4KB, which is fine as long as the server doesn't try to send a longer message. Streaming a large binary is a good way for this to happen, though!

Two ways to fix this:

* Enable the RFC 6066 max_fragment_length extension in the TLS config of the server. This is the best option, because it means client and server will negotiate a maximum 4KB message size.

* Increase the max message size (at the cost of free RAM). There is a [patch for mbedTLS that I wrote last year](https://github.com/ARMmbed/mbedtls/pull/536) that allows TX & RX buffer sizes to be different, which saves some of the RAM overhead. I should get around to looking at that again!

ole-johan, 2017-02-10:
We were able to set the server fragment length to 4096, and now it is working! 
Thanks!

Now I have a new problem, that after a number of consecutive reads within that one download, the ESP resets. But I'll try to investigate further...

closed 2017-02-22 by UncleRus

MrBuddyCasino, 2017-04-21:
Just had this issue on the ESP32 after reducing mbedTLS buffers via menuconfig - would never have figured out what was going on if it wasn't for projectgus's comment.

Maybe add a warning text to the configuration option?