SuperHouse/esp-open-rtos #336: Creating a CSR file on the ESP 8266

maximilian1000, 2017-02-08:
Hi,

do you know any example in which a CSR file is created using the mbedTLS library on the ESP 8266? Although I've looked for it a long time I didn't found a working example.

Thank you.

Max

kezl, 2017-02-10:
Are you 100% sure you need a CSR? a self signed certificate is often a better bet for an esp8266, in which case you can use openssl to generate one as in the comments in examples/tls_server/cert.c
It depends of course, what's your intended use?

pcMan101, 2017-02-10:
Thank you! But I want my ESP to generate a CSR, send it to a server which gives back a Client Certificate to my ESP.

kezl, 2017-02-10:
Ah ok, if no one else has an example you could try using the method on the mbed website:
[https://tls.mbed.org/kb/how-to/generate-a-certificate-request-csr](url)
The source for both programs mentioned is in the extras/mbedtls/ directory so it may be possible to lift the required functions from them and get it working.