SuperHouse/esp-open-rtos #344: CPP linkage for SPIFFS

MartianWearables, 2017-02-21:
It seems that cpp linkage was not considered when porting the SPIFFS module? I added the necessary 
`
#ifdef __cplusplus 
extern "C" { 
...
`
around relevant methods in the esp_spiffs.h file, which allowed it to compile.

Spiffs is able to init and mount, but `int fd = open("other.txt",  O_WRONGLY|O_CREATE, 0);` fails with "Error opening file".

Any ideas on how to get SPIFFS working from inside my cpp class?

Cheers

MartianWearables, 2017-02-22:
I solved the problem: In addition to adding the extern "C" to both included spiffs header files, I had to make sure not to use the posix "read()" or "close()" functions, and they conflicted with those provided by the lwip library. Note the the spiffs example included with esp-open-rtos does not use the SPIFFS_write(), even though it is available and functional. I think the SPIFFS example should be changed to include both posix write() and SPIFFS_write. Also, the posix file manipulation api's should be made intercompatible.

Cheers

lhartmann, 2017-03-16:
Several extras are missing that, httpd included.

I wrote a batch check/fix script. Not perfect, but does the job.
http://pastebin.com/WXfZDSQL

cd esp-open-rtos
# Get a list of headers missing extern "C" under current directory
cppchk.sh
# If you feel safe, then patch all headers
cppchk.sh --fix