SuperHouse/esp-open-rtos #345: Include extra directories

ivanroberto, 2017-02-22:
Hi, I have a project with lot of directories (.c and .h files).
Reading esp-open-rtos Wiki, in "Build Process", there is a variable "PROGRAM_SRC_DIR", that appears to solve my problem, BUT,  i have a lot of directories and i realized that only the first is working, for example in my Makefile:

`PROGRAM=terminalWifiRtos

PROGRAM_SRC_DIR = $(ROOT)examples/terminal_Wifi_Rtos/user
PROGRAM_SRC_DIR += $(ROOT)examples/terminal_Wifi_Rtos/motor
PROGRAM_SRC_DIR += $(ROOT)examples/terminal_Wifi_Rtos/display

include ../../common.mk`

So linker can only find files in "/user", for the others it failed to find my include files.

I appreciate if some one help.

Thanks

Ivan

Zaltora, 2017-02-22:
Try this:

`PROGRAM_SRC_DIR = .  ./user  ./motor ./display`

UncleRus, 2017-02-22:
I would recommend using full pathes:

```Makefile
PROGRAM_SRC_DIR = $(PROGRAM_ROOT) $(PROGRAM_ROOT)user $(PROGRAM_ROOT)motor $(PROGRAM_ROOT)display
```
Also you can manually specify all source files in your project, e.g.
```Makefile
PROGRAM_SRC_FILES = first.c subdir/second.c
```

ivanroberto, 2017-02-22:
Thanks Zaltora and UncleRus,

But still the same, it finds just the first directorie "user", if I change the order

`PROGRAM_SRC_DIR = . ./motor ./user ./display`

or

`PROGRAM_SRC_DIR = $(PROGRAM_ROOT) $(PROGRAM_ROOT)motor $(PROGRAM_ROOT)user  $(PROGRAM_ROOT)display`

then it finds only files from "motor"

But I think that may problem is another, because directories motor and display, are sub-directories of myLib
So my struct is:

examples
------------->terminal_Wifi_Rtos
----------------------------------------->user
----------------------------------------->myLib
--------------------------------------------------->motor
--------------------------------------------------->display

So I tried to do in both ways, but no luck:
`PROGRAM_SRC_DIR = $(PROGRAM_ROOT) $(PROGRAM_ROOT)user (PROGRAM_ROOT)myLib/motor $(PROGRAM_ROOT)myLib/display`

or 

`PROGRAM_SRC_DIR = . ./user ./myLib/motor ./myLib/display`

Thanks

UncleRus, 2017-02-22:
https://github.com/UncleRus/dir_test

Just tried sub-sub-dirs and it works fine

ivanroberto, 2017-02-22:
UncleRus,

Yes i ran your code and it works!
But I alredy know my problem... I tried to include the motor.h and display.h without complete path

```
#include "motor.h"
#include "display.h"
```

instead of your full path

```
#include "../myLib/display/display.h"
#include "../myLib/motor/motor.h"
```

I does this, because I saw in common.mk that use PROGRAM_SRC_DIR with `-I` prefix.

I made a small change on common.mk
Line 118
```
$(1)_CC_BASE =  $(Q)$(CC) $$(addprefix -I,$$(USER_INC_DIRS)) $$(addprefix -I,$$(PROGRAM_SRC_DIR)) $$(addprefix -I,$$(INC_DIRS)) $$(addprefix -I,$$($(1)_INC_DIR)) $$($(1)_CPPFLAGS)
```
And use a new variable "USER_INC_DIRS", that puts my diretories on path

So my Makefile become on this:

```
PROGRAM = dir_test

PROGRAM_SRC_DIR = $(PROGRAM_ROOT) $(PROGRAM_ROOT)user $(PROGRAM_ROOT)myLib/display $(PROGRAM_ROOT)myLib/motor

USER_INC_DIRS += $(PROGRAM_ROOT)myLib/motor
USER_INC_DIRS += $(PROGRAM_ROOT)myLib/display

include ../../common.mk
```

And it worked, but I don't what to do this.

Do you see another way???

Thanks

UncleRus, 2017-02-22:
No need to modify common.mk, just define `PROGRAM_INC_DIR` in your makefile right after the `PROGRAM_SRC_DIR` definition:

```Makefile
PROGRAM = dir_test

PROGRAM_SRC_DIR = $(PROGRAM_ROOT) $(PROGRAM_ROOT)user $(PROGRAM_ROOT)myLib/display $(PROGRAM_ROOT)myLib/motor
PROGRAM_INC_DIR += $(PROGRAM_SRC_DIR)

include ../../common.mk
```

ivanroberto, 2017-02-22:
Great UncleRus,

Now it works!!!

Thank you so much!!!!

closed 2017-02-22 by UncleRus