SuperHouse/esp-open-rtos #349: vApplicationIdleHook() and vApplicationTickHook()

SaimenSays, 2017-03-12:
Why is function code for `vApplicationIdleHook()` and `vApplicationTickHook()` in app_main.c?
If I want to use own hooks, I get a compiler error, because of double implementation to these of app_main.

andrewclink, 2017-03-19:
Personally I mark these as weak if I need to override them in my code. I'd be happy to submit a PR if that's an agreeable solution.

UncleRus, 2017-03-20:
> I'd be happy to submit a PR
That's would be greate!

andrewclink, 2017-03-20:
I'm writing an example at the same time to demonstrate this and would like to put the chip to sleep in the idle hook. This is just copied from the `simple` example.

Does anyone know the appropriate way of putting the chip to sleep where the FreeRTOS tick event can wake it?

andrewclink, 2017-03-21:
PR #355. I think this issue can be closed.

closed 2017-03-22 by UncleRus