SuperHouse/esp-open-rtos #351: missing c_types.h and/or stddef.h?

anyn99, 2017-03-15:
I am using your esp-open-rtos in eclipse and really like it, thanks a lot for all the work :-)

What i find strange, is that i can't find the typedef for "size_t" inside my included headers.. and i included alot:
  esp-open-rtos/include
  esp-open-rtos/FreeRTOS/Source/include
  esp-open-rtos/FreeRTOS/Source/portable/esp8266
  esp-open-rtos/libc/xtensa-lx106-elf/include
  esp-open-rtos/core/include
  esp-open-rtos/open_esplibs/include
  esp-open-rtos/lwip/include
  esp-open-rtos/lwip/lwip/src/include/ipv4
  esp-open-rtos/lwip/lwip/src/include/posix
  esp-open-rtos/lwip/lwip/src/include/lwip
  esp-open-rtos/lwip/lwip/src/include
there is a c_types.h from the original SDK at  esp-open-rtos/lwip/lwip/espressif/include, but thats somehow counter-intuitive to use that one.

What i also miss is a definition for va_list. Is it not supported yet?

UncleRus, 2017-03-15:
`size_t` defined in `<your esp-open-sdk path>/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/include/stddef.h`
You don't need to add this path manually, compiler does it. Just `#include <stddef.h>`

anyn99, 2017-03-15:
Duh, didn't think about the toolchain-includes since I didn't configure the toolchain in Eclipse!
Thanks a lot!

closed 2017-03-15 by UncleRus