SuperHouse/esp-open-rtos #354: Idle Task never runs

EFudd, 2017-03-20:
Hi I am working on a project based on the HTTP_SERVER websocket example.  The IDLE task never runs so deleted tasks are not cleaned up, hence the server crashes after a few new connections because it runs out of stack space.

SaimenSays, 2017-03-20:
I had same Issue some time ago and it turned out, that uart read blocks. So the task which calls uart function gets never suspended and RTOS never runs Idle.
So I think you have a task with a loop causing 100% CPU load, so IdleTask will never switched in.

EFudd, 2017-03-21:
Thanks for the rapid response:  UART I/O on these devices causes lots of 
WDT problems in all SDK's so my RTOS code is peppered with 
wTaskDelay(0)'s around anything that looks like it could block, 
especially I/O.  All task switching works except for IDLE never 
running.  As an experiment I upped the priority of the IDLE task to 2 to 
match the other lowest priorities and it seems to work properly:  IDLE 
runs and cleans up deleted tasks.  Obviously, this is not the proper 
solution, but maybe a clue.  My diagnostic output is on UART1.  I use 
UART0 in inverse mode to read from a cheap RS232 camera.  The camera 
works fine and I/O with it is only under a trigger, that is, even when  
no I/O on UART0 has been initiates the IDLE problem is still there.    
Digging into the RTOS code is fun, but I couldn't get to the core of the 
scheduler.  Is all or part of it inside the ESP ROM code?

Regards E.F.

EFudd, 2017-03-21:
Hi, with my IDLE not running problem.  The HTTP deamon in the 
HTTP_SERVER example has a hard loop.  I put vTaskDelay(0) in the loop  
==> taskYIELD().  I did not realize that the FREERTOS yield is only to a 
task of equal or higher priority.  Using vTaskDelay(1)  breaks the HTTP 
deamon's grip and allows IDLE to run.  By the WAY, problem exists in the 
HTTP_DRIVER example.

Thanks much, E. F,.

PS When finished, I will post some OPEN RTOS examples somewhere for the 
public:  A better ds18b20 example that uses interrupts on the onewire 
signal and voids wasting cpu time, and the http server with a RS232 
camera on UART0@115200.  The other RS232 examples I have seen us a 
software uart, are slow and unreliable.

kanflo, 2017-04-12:
Looking forward to your onewire IRQ example.

closed 2017-04-12 by kanflo

billiumbuck, 2017-06-03:
I know this is closed but wouldn't putting taskyield or taskdelay int the forever loop of httpdtask help here?