SuperHouse/esp-open-rtos #359: hooks_example

SaimenSays, 2017-03-25:
Hi!

I think use of `sdk_wifi_set_sleep_type()` in IdleHook is wrong. According to [Espressif docu](http://www.espressif.com/sites/default/files/9b-esp8266-low_power_solutions_en_0.pdf) this function is for automatic sleep and does not need to be called in each tick!
Also if I try this, ESP crashes because of printing "sleep enable,type: 2" in each tick.

andrewclink, 2017-03-27:
The sleep call might fail because wifi isn't active. I really didn't know how to implement sleep within esp-free-rtos, but I'd love to learn!

Zaltora, 2017-04-03:
something like this work better ? 
```
    if (sdk_wifi_get_sleep_type()  !=  WIFI_SLEEP_MODEM)
        sdk_wifi_set_sleep_type(WIFI_SLEEP_MODEM);
```
didn't know how work sleep system with esp8266.

SaimenSays, 2017-04-03:
Even if sleep type is set in my project, I can't measure any lower current consumption. So I doubt it is working properly from ESP side. So maybe all discussions are needless.
Can somebody measure it again and tell the results....

resetnow, 2017-04-03:
>I can't measure any lower current consumption

as in you can't measure current at all or the measured current doesn't change? Out of curiosity, what are your results?

SaimenSays, 2017-04-03:
Compiled, reprogrammed and runned it twice, one time without sleep and second time witth sleep. No WLAN traffic to ESP after startup. The measured current differs less then noise. It is not necessary to set it if it does not improve anything.
Of course it would be nice to save some milliwatts in standby...

Zaltora, 2017-04-04:
With standard sdk, we need restart device after set sleep type ?
`void sdk_system_restart(void);` ( "include/espressif/esp_system.h")

I found these functions too:
`bool sdk_system_restart_enhance(uint8_t bin_type, uint32_t bin_addr);` "include/espressif/user_interface.h"
`void sdk_system_restart_in_nmi(void);` "open_esplibs/include/esplibs/libmain.h"

Zaltora, 2017-04-05:
I do some test to disable Wifi part. I added 
```
    sdk_wifi_station_stop();
    sdk_wifi_softap_stop();
```
Don't put it in `user_init(void)`, because in app_main.c it will be enable again by default after your initialization. Esp8266 will stop current spike. It still consume 0.07A.
I found these functions too in this file "lib/allsymbols.rename":
```
    sdk_wDevDisableRx();
    sdk_wDev_DisableTransmit();
    sdk_phy_disable_agc();
```
but the first two are not defined. I am not sure if they are usefull but i cant test it. I try to use binary sdk but i got more undefined functions.
warning: implicit declaration of function 'sdk_wDevDisableRx'
warning: implicit declaration of function 'sdk_wDev_DisableTransmit'

paulmthompson, 2017-06-09:
I have been looking at how sleep in esp-open-rtos compares to the official espressif RTOS SDK and there are a few differences.

In the official SDK, you don't need an idle hook to make modem sleep work. You instead configure the sleep type to be modem, and then force sleep. In the RTOS version, even though sleep is "forced", it should only begin whenever the idle task is being executed. Consequently, I don't think you actually need to call anything from the idle task hook. The call would look like this:
```
wifi_set_sleep_type(MODEM_SLEEP_T) //Sleeps between wifi beacons;
wifi_fpm_open() //Necessary to use force sleep functions;
wifi_fpm_do_sleep(0xFFFFFFFF) //Force sleep for time in microseconds. Maximum value indicates that wifi will sleep until an explicit wakeup call is encountered (wifi_fpm_do_wakeup()). Doesn't start until idle task is excecuted;
```

Somewhat confusingly in their documentation, their example also disconnects from the WIFI network with the stop functions. This kind of defeats the purpose of modem sleep though, where you are sleeping between wifi beacons, so you can remain connected to the network, just not transmit or receive information during that time.

For this SDK, I don't think there are enough implemented functions currently to use modem sleep in this way. The "lib/allsymbols.rename" has a placeholder for the `sdk_pm_open` function, as well as `sdk_pm_open_rf` (not sure which one is the same as `wifi_fpm_open()`). There is no `sdk_pm_do_sleep`, but there are several ones that look like that, like `sdk_pm_prepare_to_sleep`, `sdk_pm_set_sleep_time`, and `sdk_pm_sleep_for`.

So in summary:

1) The idle hooks does work, but I don't think the modem sleep, even if working, would be the best example. Maybe instead have something where a counter variable is incremented when it enters the hook and a normal task prints that variable after a delay to show that it was successfully incremented? I can confirm that this works and would be happy to do a pull request.

2) Playing around with this example convinced me that modem sleep is going to require a little more digging in the pm.o library and implement a few more functions.

Hope this helps!