SuperHouse/esp-open-rtos #362: redirect STDOUT to U1TXD

emichel11, 2017-04-01:
I tried to redirect stdout (ie printf etc) to U1TXD.  In core/app_man.c the output is defaulted to "sdk_os_install_putc1(default_putc)" which is U0TXD.  
I'ved tried to change that in app_main as well as in my app by calling the function "sdk_os_install_putc1" -which is in fact os_install_putc1- and set it to "uart_putc(1, c);"
However output still remains on UART0.   
Any idea what is wrong or what it is needed to change??

resetnow, 2017-04-01:
Have you tried calling `set_write_stdout`?

emichel11, 2017-04-02:
Thanks resetnow, I did not know this function.

emichel11, 2017-04-03:
To redirect STDOUT from UART0 to UART1 one need to call
PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO2_U, FUNC_U1TXD_BK);
first.
Afterwards a call of "set_write_stdout" is successful. The function template is found in   
core/newlib_syscalls.c

closed 2017-04-03 by emichel11