SuperHouse/esp-open-rtos #364: odd SNTP behavior that sometimes happens when synchronizing

jkent, 2017-04-14:
My project uses the SNTP code -- most of the time it behaves as expected, but occasionally it will act up and time will go backwards a few hours a second, starting from the synchronized time.  Here is my code I've tested it with:

https://gitlab.jkent.net/wtf/wifilcd/blob/master/main.c#L53
https://gitlab.jkent.net/wtf/wifilcd/blob/master/clock.c#L245

and here is the observed behavior:

https://www.youtube.com/watch?v=eYdcxB7kLnQ

kanflo, 2017-04-18:
That's odd. Have you verified the SNPT replies with printf's or eg. wireshark?

jkent, 2017-04-18:
Not yet, I should have time this week to dig in deeper.

Rutger798, 2020-04-15:
I have the same issue, and enabled some extra SNTP logging with: -DSNTP_LOGD -DSNTP_LOGD_WITH_PRINTF
after some day's of logging it looks like "RTC counter wrapped" occurred

And that only happens in: 
`
// Check if a timer wrap has occurred. Compensate sntp_base reference
// if affirmative.
// TODO: think about multitasking and race conditions
static inline void sntp_check_timer_wrap(uint32_t current_value) {
	if (current_value < tim_ref) {
		// Timer wrap has occurred, compensate by subtracting 2^32 to ref.
		sntp_base -= 1LLU<<32;
		// DEBUG
		SNTP_LOGD("SNTP RTC counter wrapped");
	}
}
`

and that also shows the issue in the comment, "multitasking"...

Rutger798, 2020-04-15:
Soo that brings me to the open pull requests:
But both are to old to merge.....

https://github.com/SuperHouse/esp-open-rtos/pull/157
https://github.com/SuperHouse/esp-open-rtos/pull/148