SuperHouse/esp-open-rtos #368: question about esp-open-sdk version

marduino, 2017-04-17:
Hi,

Do I need to build esp-open-rtos with modified version of "VENDER_SDK" from "esp-open-sdk/Makefile"?
Since the latest version set it as 2.0.0. 

1. I tried with default built, it will use "ESP8266_NONOS_SDK_V2.0.0_16_08_10.zip" by default, I tried some program, works fine until now.
2. I cann't find a version of 0.9.9. If I have to specify a version, which one should I use?

I read from this wiki page "https://github.com/SuperHouse/esp-open-rtos/wiki/ESP-SDK-Differences", it says that 
> "For licensing reasons, it is based on Espressif RTOS SDK 0.9.9 rather than newer versions of the SDK (where the license changed). This means that some of the newer features in Espressif's SDK are not found in esp-open-rtos."

But in the README.md, it didn't mention this:

> Install esp-open-sdk, build it with make toolchain esptool libhal STANDALONE=n, then edit your PATH and add the generated toolchain bin directory. The path will be something like /path/to/esp-open-sdk/xtensa-lx106-elf/bin. (Despite the similar name esp-open-sdk has different maintainers - but we think it's fantastic!)

@Angus71

resetnow, 2017-04-17:
esp-open-sdk != Espressif SDK. 

esp-open-rtos includes libraries and some files from Espressif SDK 0.9.9. esp-open-sdk is a collection of tools and libraries for bootstrapping a build environment for esp8266. Espressif SDK in esp-open-rtos provides hardware-specific code mostly related to WiFi.

jkent, 2017-04-17:
What this means to you: if you're using esp-open-rtos, it only uses esptool.py and the built toolchain from esp-open-sdk

marduino, 2017-04-18:
Oh, looks complex, I am confused:
1. The version 0.9.9 means libraries under "esp-open-rtos/lib", --> then, where are they included in esp-open-sdk? How to get/upgrade them? If possible to use a later version?
2. Does "ESP8266_NONOS_SDK_V2.0.0_16_08_10.zip" = "Espressif SDK"? If yes, will any potential risk of building esp-open-sdk with such a version?

resetnow, 2017-04-18:
>where are they included in esp-open-sdk

As far as I know, they aren't included in esp-open-sdk.

>If possible to use a later version

You can try but things will very likely break.

marduino, 2017-04-26:
Hi @resetnow,
Sorry for the late response, I was out for bussiness.

But my confusion is not answered, can you explain clearly?

GBinWin, 2017-08-07:
Use espressif official release lubuntu compiled esp-open-sdk has been unsuccessful, I hope to use esp-open-rtos, a good environment or a very detailed environment to build the document , do not understand linux, windows use.