SuperHouse/esp-open-rtos #371: Not able to use GPIO-16 for ESP-12 board?

rameshKrSah, 2017-04-23:
Hello
I am trying to use the GPIO-16, of ESP8266 exposed in the ESp-12 module, but I am not able to use it.

Looking in the iomux_reg.h I can only find 0 to 15 pins listed. I am not sure what to do?

resetnow, 2017-04-23:
GPIO 16 is connected to a different hardware module. gpio_* functions handle it as a special case https://github.com/SuperHouse/esp-open-rtos/blob/master/core/include/esp/gpio.h#L82

rameshKrSah, 2017-04-24:
@resetnow thank you for the reply and I have tried using the functions from the gpio.h file. I first enabled the Pin as _GPIO_OUTPUT _ and tried to toggle it, but it didn't work. 

This the code I am using:

`gpio_enable(16, GPIO_OUTPUT);
while(1){
  gpio_write(16, 0);
  vTaskDelay(5000);
 gpio_write(16, 1);
 vTaskDelay(5000);
`

Also, this is the pin out of the ESP-12 module I am using: https://drive.google.com/open?id=0B2dAT76_2LrzXy1TTC1NNHdocmc

resetnow, 2017-04-24:
vTaskDelay(5000) is a 50 second delay, are you sure you wanted to wait that long? Anyway, it should work. Before gpio16 support was added to esp-open-rtos, I was using this code:

```
static void gpio16_output_conf() {
    RTC.GPIO_CFG[3] |= BIT(0);
    RTC.GPIO_CONF &= ~BIT(0);
    RTC.GPIO_ENABLE |= BIT(0);
}

static void gpio16_output_set(uint8_t value) {
    if (value) {
    	RTC.GPIO_OUT |= BIT(0);
    } else {
    	RTC.GPIO_OUT &= ~BIT(0);
    }
}
```

rameshKrSah, 2017-04-26:
@resetnow Thank you, and vTas

closed 2017-04-26 by rameshKrSah