SuperHouse/esp-open-rtos #376: linker error with LIBS=m before including common.mk

briantilley, 2017-05-05:
I need the `sqrt()` function from math.h, so I added `#include <math.h>` to my source and `LIBS=m` to my Makefile before `include <path_to_esp-open-rtos>/common.mk`. 

compiler output:
`CC /home/brian/Desktop/wifi-pwm/user_main.c
AR build/program.a
LD build/wifi-pwm.out
/opt/Espressif/esp-open-rtos/ld/program.ld:21: undefined symbol `malloc' referenced in expression
collect2: error: ld returned 1 exit status
/opt/Espressif/esp-open-rtos/common.mk:212: recipe for target 'build/wifi-pwm.out' failed
make: *** [build/wifi-pwm.out] Error 1`

Makefile:
`PROGRAM=wifi-pwm
PROGRAM_SRC_DIR=.
ESPBAUD=454800
EXTRA_COMPONENTS=extras/pwm extras/mbedtls extras/httpd
LIBS=m
EXTRA_CFLAGS=-DLWIP_HTTPD_CGI=1 -I./fsdata -mlongcalls -mtext-section-literals
include /opt/Espressif/esp-open-rtos/common.mk`

I have since rebuilt the entire toolchain from scratch, making sure to build esp-open-sdk with `make toolchain esptool libhal STANDALONE=n` as suggested by issue #79, and the problem persists. Are there any other libraries or flags I'm missing?

Zaltora, 2017-05-05:
to include library math, i doing this in my project:
`LIBS = hal gcc c m`

Zaltora, 2017-05-05:
In parameters.mk we found this:
`LIBS ?= hal gcc c`
if you write 
`LIBS=m`, 
it will replace LIBS variable

briantilley, 2017-05-05:
Thank you! I've issued #377 to make it easier for future users. It seems to me there should be a way to simply add libraries per-project without concern for parameters.mk.