SuperHouse/esp-open-rtos #378: sdk_wifi_station_get_connect_status() returns 255 constantly

xarrange, 2017-05-06:
Hi guys, how's that possible? There is no such status.
I tried default http_get demo.
Any ideas? Is it related to phy calibration data?

UncleRus, 2017-05-13:
Hi!
It returns 0xff in two cases:
- current opmode is `SOFTAP_MODE` (`sdk_wifi_get_opmode()==SOFTAP_MODE`)
- for some reason netif descriptor is NULL (`sdk_g_ic.v.station_netif_info==NULL`)
I think it can be related with errors in phy claibration data.

rameshKrSah, 2017-08-15:
@xarrange you can check the function definition for **sdk_wifi_station_get_connect_status()** in **user_interface.c**, to dig deeper. I just added my own custom function to return the wifi_connect_status and it worked fine for both dynamic and static IP allocations.

```
uint8_t get_wifi_connect_status(void){
	return sdk_g_ic.v.station_netif_info->connect_status;
}
```

Also, after I added this function and used it, function **sdk_wifi_station_get_connect_status()** also started to return correct values. I don't know what happened, but now I am using **sdk_wifi_station_get_connect_status()** and I am getting the right status. I know this is weird, but this is what worked for me.

apiel, 2018-02-05:
"for some reason netif descriptor is NULL (sdk_g_ic.v.station_netif_info==NULL)" how to fix such issue?