SuperHouse/esp-open-rtos #379: I2C HD44780 does not work

zgoda, 2017-05-10:
The example for HD44780 with I2C adapter does not work for me. The display is OK and works with ESP Arduino core or Sming using F. Malpartida's LiquidCrystal_I2C. Looking for differences, there is one pin that is not defined in esp-open-rtos code, namely "rw", connected to pin 1 in I2C adapter.

Definition in esp-open-rtos:

```C
	hd44780_t lcd = {
		.addr = 0x3f,
		.font = HD44780_FONT_5X8,
		.lines = 2,
		.pins = {
			.rs = 0,
			.e  = 2,
			.d4 = 4,
			.d5 = 5,
			.d6 = 6,
			.d7 = 7,
			.bl = 3
		},
		.backlight = true
	};
```

And using LiquidCrystal:

```C++
LiquidCrystal_I2C lcd(0x3f, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);
```

Pin order in constructor: en, rw, rs, d4, d5, d6, d7, backlight and apart of missing rw they're exactly the same as for esp-open-rtos lib.

resetnow, 2017-05-10:
RW pin should be tied to logical 0. When pcf8574_port_write is called, it's not set explicitly so I assume it's 0 there.

Are there any changes on display pins at all? Could it be related to I2C frequency?

zgoda, 2017-05-10:
Physically same display unit and board, no changes to wiring (SCL = 5, SDA = 4). I2C frequency not touched so I assume default.

Zaltora, 2017-05-11:
try to replace in i2C lib these value.  It will change frequency  ( <100KHz )
```
#define I2C_CUSTOM_DELAY_160MHZ   200
#define I2C_CUSTOM_DELAY_80MHZ    40
```

zgoda, 2017-05-11:
That did not change anything.

UncleRus, 2017-05-13:
@zgoda What chip is used on your adapter? Is it pcf8574?

zgoda, 2017-05-15:
Yes, exactly.

nopnop2002, 2019-01-28:
I tried examples/i2c_lcd_test.
Everything work fine without changing the cord.
I used PCF8574P, and VCC,A0,A1,A2 is 3.3V.
So the i2c address is 0x27.