SuperHouse/esp-open-rtos #386: Improve softuart implementation.

hetii, 2017-05-30:
Hello :)

I just play a bit with softuart library from extras and notice that is very error prone.

So I put softuart_put() in between  taskENTER_CRITICAL();  and taskEXIT_CRITICAL();
and it seams that help a lot. Error still happens but not so often without it.

If you have any other idea how we could improve softuart implementation feel free and share your ideas.

UncleRus, 2017-05-30:
Hi @hetii !
That's great! Would you please create the pull request?

hetii, 2017-05-30:
Sure as soon as I do more tests and add more common operating modes.

shaag7967, 2017-06-11:
had the same problems... and basically the same solution. If you neen some inspiration: [softwareUart](https://github.com/shaag7967/espNode1/blob/SDS021-and-DHT/src/softwareUart.c)