SuperHouse/esp-open-rtos #392: Newbie Question: Why can't I get useful data back over my UART terminal?

badotter, 2017-06-11:
Hi, sorry, total newb here, I'd go to the forums rather than posting an issue if I knew where to find them... anyway, I'm running ubuntu, I've been experimenting with esp-open-sdk for some days now, and I just discovered this esp-open-rtos repo today. 

I can (apparently) build and flash the example projects successfully, in both open-sdk and open-rtos, however my problem is that when I attempt to connect to the chip with my USB UART cable via any terminal application (I have tried putty and screen so far) I get nothing but a stream of garbage characters. I've tried many different baud rates, mostly 115200 but also every other one I've seen anyone mention in all my googling. I can communicate with an unflashed factory stock chip at 115200, but anything I have flashed onto the chip results in constantly blinking blue LED and constant stream of garbage to my terminal.

Am I missing some totally obvious step? It is possible that my test chip is fried, I have more chips coming in the mail any day now, but my tester does seem to behave fine when it comes to flashing at least. Any advice would be welcome! Thanks for all of your hard work.

resetnow, 2017-06-11:
Hello,

you can try calling uart_set_baud or running <a href="https://github.com/SuperHouse/esp-open-rtos/blob/master/examples/serial_echo/serial_echo.c#L13">serial_echo</a> example.

badotter, 2017-06-11:
AH, thank you so much! I don't know what my problem is, but serial_echo worked!!! We got boots on the ground.

closed 2017-06-12 by badotter