SuperHouse/esp-open-rtos #395: Make works for blink_timers but not for blink example

ghislainborremans, 2017-06-13:
When i make the blink_timers example, evrything ok
when i MAKE the blink example the process stops because it cannot find a file 
Reason: the path is not correct: it contains 2 times esp-open-rtos: see below
home/ghislain/ESP8266/esp-open-rtos/esp-open-rtos/open_esplibs/libmain/user_interface.c

I cannot find where i could change this or where i have put a parameter that causes this? 
any ideas where to look? 

best regards

resetnow, 2017-06-14:
Can you paste terminal output here?

ghislainborremans, 2017-06-14:
Feeling very stupid. In the process of making the terminal output , i first made a "make clean" in order to have a good starting point.=>> the problem has disappeared.!!
Thank you for your speedy reply.

closed 2017-06-14 by ghislainborremans