SuperHouse/esp-open-rtos #396: Can anything be moved from iram to rom?

anyn99, 2017-06-18:
Hi there,

i integrated a custom memory allocator (umm_malloc) into esp-open-rtos.
But since it has to be in iram, i can't find enough space for it!
Is there anything i could remove from iram?
Of course i already removed mallocr.o and freer.o, since i replace them completly.

Objects i tried to move to rom so far: 
*libgcc.a:*div*i3.o -> fails when changing wifi credentials
*libc.a:*rand.o -> fails when used
i don't think it makes sense to touch any of the mem* functions, does it?

I really only need a few hundred bytes more, can you help me?

anyn99, 2017-07-13:
Hi again,

I'm here to answer my own question, in case anybody will ever need it.
The reason why the libgcc *i3 functions can't be moved to flash is not because they are called before flash is mapped. It's rather that they are called inside ISRs and the flash can be unmapped during this.
The guys from the esp8266-Arduino-core experienced something similar:
e.g. https://github.com/esp8266/Arduino/issues/1403
You get exception 0 (illegal instruction) when that happens.

My solution is now to take the PROVIDE() statements from rom.ld that contain the basic gcc arithmetic functions:
```
PROVIDE ( __adddf3 = 0x4000c538 );
PROVIDE ( __addsf3 = 0x4000c180 );
PROVIDE ( __divdf3 = 0x4000cb94 );
PROVIDE ( __divdi3 = 0x4000ce60 );
PROVIDE ( __divsi3 = 0x4000dc88 );
PROVIDE ( __extendsfdf2 = 0x4000cdfc );
PROVIDE ( __fixdfsi = 0x4000ccb8 );
PROVIDE ( __fixunsdfsi = 0x4000cd00 );
PROVIDE ( __fixunssfsi = 0x4000c4c4 );
PROVIDE ( __floatsidf = 0x4000e2f0 );
PROVIDE ( __floatsisf = 0x4000e2ac );
PROVIDE ( __floatunsidf = 0x4000e2e8 );
PROVIDE ( __floatunsisf = 0x4000e2a4 );
PROVIDE ( __muldf3 = 0x4000c8f0 );
PROVIDE ( __muldi3 = 0x40000650 );
PROVIDE ( __mulsf3 = 0x4000c3dc );
PROVIDE ( __subdf3 = 0x4000c688 );
PROVIDE ( __subsf3 = 0x4000c268 );
PROVIDE ( __truncdfsf2 = 0x4000cd5c );
PROVIDE ( __udivdi3 = 0x4000d310 );
PROVIDE ( __udivsi3 = 0x4000e21c );
PROVIDE ( __umoddi3 = 0x4000d770 );
PROVIDE ( __umodsi3 = 0x4000e268 );
PROVIDE ( __umulsidi3 = 0x4000dcf0 );
```
then i removed the PROVIDE() to make them fixed assignments and put them at the start of program.ld.
This way the built in rom functions replace the ones from libgcc. I think they are simple enough to assume that the are moreor less like the current gcc ones.
Here we go,i have around 2kb more iram for my stuff!

Another approach would have been to completly replace rand and srand by putting hwrand in iram and wrapping rand calls to hwrand. We anyway initialize srand with hwrand. 
The onlydrawback would be, that you can't get a fixed pseudorandom sequence from rand anymore. But who needs this anyway?

closed 2017-07-13 by anyn99

ourairquality, 2017-07-13:
Great, and extra 2k is a good win, thanks for reporting this.

fwiw I expect in the current code that an ISR can run in flash. e.g. It appears that the wdev interrupt handler calls malloc which is in flash. Note the comment in the link that this is ok so long as all uses of the spi disable all interrupts, and I think the esp-open-rtos code does that in critical sections and wraps spi uses in critical sections - if not then that might need to be explored?

anyn99, 2017-07-13:
Hmm, interesting. I actually didn't remove all *i3 functions from iram, just the *div* functions, as they need the most memory and are not necessary (yet) before flash is mapped.
To trigger this i had to rebuild completly and erase the flash first, otherwise __div is not called (seems to be only needed for some first time wifi connection stuff).
I want to note that my setup is partly modified esp-open-rtos, but none of my changes should be causing that i think.
The device starts normally until i want to connect to AP:
```
mode : sta(60:01:94:16:e9:89)
add if0
Fatal exception (0):
epc1=0x4024f228

0x4024f228: __udivsi3 at /cygdrive/d/Daniel/git/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libgcc/config/xtensa/lib1funcs.S:515

epc2=0x00000000
epc3=0x00000000
excvaddr=0x40261368

0x40261368: blowfish_enc at blowfish.c:?

depc=0x00000000
excsave1=0x401006a5

0x401006a5: sdk_SPIEraseSector at /cygdrive/d/Daniel/git/esp-open-rtos/open_esplibs/libmain/spi_flash.c:102

Registers:
a0 401006a5 a1 3fffa000 a2  00400000 a3  00001000

0x401006a5: sdk_SPIEraseSector at /cygdrive/d/Daniel/git/esp-open-rtos/open_esplibs/libmain/spi_flash.c:102

a4  60000200 a5  fffdffff a6  04000003 a7  fffffffe
a8  04000002 a9  04000003 a10 00000001 a11 04000002
a12 3ffe88a8 a13 000003fd SAR 00000016

Stack: SP=0x3fffa000
0x3fffa000: 003fd000 3fffad6c 000003fd 401007dc

0x401007dc: sdk_spi_flash_erase_sector at /cygdrive/d/Daniel/git/esp-open-rtos/open_esplibs/libmain/spi_flash.c:157

0x3fffa010: 40100878 00000001 00000370 402625e6

0x40100878: sdk_SPIRead at /cygdrive/d/Daniel/git/esp-open-rtos/open_esplibs/libmain/spi_flash.c:94
 (inlined by) sdk_spi_flash_read at /cygdrive/d/Daniel/git/esp-open-rtos/open_esplibs/libmain/spi_flash.c:187

0x402625e6: sdk_system_get_checksum at ??:?

0x3fffa020: 40262679 00001000 3ffea63c 4025d630

0x40262679: sdk_wifi_param_save_protect at ??:?

0x4025d630: mbedtls_x509_time_is_future at ??:?

0x3fffa030: 402501e7 000003fd 3ffea63c 4026265c

0x402501e7: strncmp at ??:?

0x4026265c: sdk_wifi_param_save_protect at ??:?

0x3fffa040: ffffff00 ffffffff ffffffff ffffffff
0x3fffa050: ffffffff ffffffff ffffffff 000003fc
0x3fffa060: 000003ff 4026252c 00000001 00000000

0x4026252c: sdk_wifi_set_opmode_local at ??:?

0x3fffa070: 00000001 40262552 3fffa4a0 3ffea5c4

0x40262552: sdk_wifi_set_opmode at ??:?

Free Heap: 52464
_heap_start 0x3fff2e10 brk 0x3fff2e10 supervisor sp 0x3ffffb00 sp-brk 52464 bytes
arena (total_size) 0 fordblks (free_size) 0 uordblocks (used_size) 0
```

anyn99, 2017-07-13:
Please ignore all heap related output, i use my own allocator so mallinfo() is reporting crap :-)

anyn99, 2017-07-13:
And here another one when commenting out " *libc.a:*rand.o(.literal .text .literal.* .text.*) " from iram section. This one triggers after a lot of tcp traffic and is quite indeterminant, which points to some race condition involving interrupts?
```
Fatal exception (0):
epc1=0x40214534

0x40214534: rand at /home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/rand.c:84

epc2=0x00000000
epc3=0x401010b6

0x401010b6: read_block at /cygdrive/d/Daniel/git/esp-open-rtos/core/spiflash.c:182
 (inlined by) read_data at /cygdrive/d/Daniel/git/esp-open-rtos/core/spiflash.c:204
 (inlined by) spiflash_read at /cygdrive/d/Daniel/git/esp-open-rtos/core/spiflash.c:225

excvaddr=0x4024e944

0x4024e944: tcp_poll at ??:?

depc=0x00000000
excsave1=0x40102e71

0x40102e71: sdk_lmacTxFrame at ??:?

Registers:
a0 40102e71 a1 3ffea1a0 a2  00000000 a3  0000000d

0x40102e71: sdk_lmacTxFrame at ??:?

a4  0142f31c a5  3ff20a00 a6  00060000 a7  00000000
a8  3ff209d0 a9  0006d274 a10 00060000 a11 00000005
a12 3ffec8c8 a13 3ffec850 SAR 00000020

Stack: SP=0x3ffea1a0
0x3ffea1a0: 00000005 40102a47 3ffec8c8 3ffeba64

0x40102a47: sdk_lmacMSDUAged at ??:?

0x3ffea1b0: 00000001 40102531 3ffec8c8 04000000

0x40102531: sdk_lmacRecycleMPDU at ??:?

0x3ffea1c0: 4010478b 3ffeba64 00186a00 00000000

0x4010478b: sdk_trc_NeedRTS at ??:?

0x3ffea1d0: 4010294b 3ffec8c8 3ffeb6e0 0000007f

0x4010294b: sdk_lmacMSDUAged at ??:?

0x3ffea1e0: 3ff20a00 00040000 00000000 00000000
0x3ffea1f0: 40102c2d 00080000 00000002 00000100

0x40102c2d: sdk_lmacProcessAckTimeout at ??:?

0x3ffea200: 40104ecc 40104db1 3ff20a00 00000000

0x40104ecc: sdk_wDev_ProcessFiq at ??:?

0x40104db1: sdk_wDev_ProcessFiq at ??:?

0x3ffea210: 00000000 53002200 04e62528 401002bc

0x401002bc: NMIExceptionHandler at exception_vectors.o:?

Free Heap: 52464
_heap_start 0x3fff2e10 brk 0x3fff2e10 supervisor sp 0x3ffffb00 sp-brk 52464 bytes
arena (total_size) 0 fordblks (free_size) 0 uordblocks (used_size) 0

```

ourairquality, 2017-07-14:
Interesting, so the flash code wraps uses of the spi in `vPortEnterCritical` and `vPortExitCritical` but that must only disable the maskable interrupts. The wdev interrupt seems to be a NMI, so must still run, and thus all those paths might need to be in iram. Since we don't seem to have malloc in iram at present and the wdev handler calls malloc that seems a problem that needs fixing.

You might be able to test that by also disabling these NMIs when using the spi. For example try these:
```
void IRAM vPortEnterCritical(void) {
    if (!sdk_NMIIrqIsOn) {
        portDISABLE_INTERRUPTS();
        do {
            DPORT.DPORT0 &= 0xFFFFFFE0;
        } while (DPORT.DPORT0 & 1);
    }
}

void IRAM vPortExitCritical(void) {
    if (!sdk_NMIIrqIsOn) {
        DPORT.DPORT0 = (DPORT.DPORT0 & 0xFFFFFFE0) | 1;
	portENABLE_INTERRUPTS();
    }
}
```

Unfortunately wDev_ProcessFiq needs work, it just asserts if a frame was missed, but see how it goes and that can be fixed too. The wdt might also trigger on long spi operations, and that can be fixed too.

Another option might be to audit those paths, work out a minimal amount of code that needs to be in iram for the wdev handler and ensure it is. It seems odd to allocate the rx esf buffers from a pool but to be calling malloc for the small pp_post objects, and perhaps such allocations could be from a pool too.

anyn99, 2017-07-14:
I can test that, but it seems odd to me delaying the lowlevel wifi handling with every flash access.
I think it is a better option to keep needed code in iram. And I am not even sure that malloc is placed in flash? i thought i saw it mapped to iram the other day, maybe some linker/compiler optimization?
I really don't know much about the wdev stuff yet, so i only partially understand what you are saying :-)
Especially the last bit about pp_post objects is cryptic for me. ;-)

ourairquality, 2017-07-14:
@anyn99 Yes, I guess it would be best to keep all that path in iram.

Looking again I think you are right that malloc is in iram, sorry.

Taking another look at the pp_post paths, perhaps it does allocate from a pool too, sorry. The func002 might be allocating two word objects from a pool for pp_post and filling these with the first and second arguments for pp_post, and pp_task calls func012 at the end of each loop which might free from this pool. 

The first argument to pp_post seems to be a code and for 0,1,2,3 it used xQueueSendToBack and these appear to for TX paths so probably not from the interrupt handler. 

Code 5 looks like the normal RX interrupt path, but rather than just sending to the queue, it allocates an object using func003 than calls `PendSV(2)` which might be effectively portYield. This in turn appears to end up at MacIsrSigPostDefHdl which calls func002 and moves the entry to that queue. This path does seems really odd. Why go to all that trouble?

I don't yet understand how all this fits together, but I think a good command of this code is needed.

ourairquality, 2017-07-20:
If only for the record, hit a use of malloc in the NMI - lmac calls rand() and the newlib version uses the reent structure which is dynamically allocated. Even using a reent struct in the NMI is wrong. I think we can fix that by renaming rand() in the sdk and supplying a safe version.