SuperHouse/esp-open-rtos #398: Linker Issue

gpreviato, 2017-06-22:
Hi, I have this kind of matter, and I'd like to know if there is a 'standard' way to approach it.

Due to some problems on boot, I'm partially rebuild the code of the phy_chip_v6 module. 
In the register_chipv6_phy the original routine calls an internal static struct, defined in the .bss segment, just local, that is obviously relocated on linking. No way to access it by name.

But I know the indexes in the relocation table and the 'value' in the relocation word.

I need to 'connect' this structure (for initialization), and the only way I have found is to modify the new object changing the relocation references in, that is a bit struggling...

I wonder if there are links commands, scripting or other 'tricks' to manage it in a better way, specially to have a smooth compilation more compliant with the actual one...

Many thanks,

--gra

ourairquality, 2017-06-22:
Not sure how you can do what you ask, but there appear to be enough known labels to uses as a based to get to other offsets. The argument to `sdk_register_phy_ops` looks like a function table that might be static and could be reproduced. Is there any reference there that is a problem?

gpreviato, 2017-06-22:
For my understanding, the argument of sdk_register_phy_ops is chip_v6_rf_init, but could be I'm wrong... 
Actually I'm stuck on another internal variable, used to read and write, that I'm unable to refer to any other. So I was wondering if I could just 'link' to it....

ourairquality, 2017-06-22:
@gpreviato I don't know how to link to static variables, but is their a label before this that you can use a base? What is the offset from `rxiq_compute_num` of the location that is a problem?

gpreviato, 2017-06-22:
If you give a glance to the disassembly, at offset 3864 there is a l32r that is loading an address into 15. 
The content of this mem position (initialized as 0) is checked internally in different positions and just changed to 1 before to end from the function.
Seems just a flag, to avoid that the same code is performed more than one time, but could be not...

gpreviato, 2017-06-22:
Sorry, I'm a bit tired... find out now the right offset! Many thanks for your suggestion!