SuperHouse/esp-open-rtos #399: Many compiler errors

tho068, 2017-06-29:
Hello

I am trying to 'make flash' one of the examples that is bundled with the esp-open-rots system. I have compiled the SDK, installed esptool etc.

[debug.txt](https://github.com/SuperHouse/esp-open-rtos/files/1111561/debug.txt)

The output from the compiler is in the attatched file.

I suspect that my compiler is too new or something.

Anyone know what is going on here?

ourairquality, 2017-06-29:
@tho068 It should build fine with the latest esp-open-sdk, and the travis builds use a recent version. It look like it is picking up the wrong include files. The esp-open-sdk can be build in different ways, and perhaps double check that it was built as requested for esp-open-rtos, README.md.

tho068, 2017-06-29:
Thanks.

--recursive on git clone :-)

closed 2017-06-29 by tho068