SuperHouse/esp-open-rtos #400: How to get rid of all system UART messages?

xarrange, 2017-07-03:
Is it possible to turn off system printouts like scandone, reconnect, add, etc... ?
E.g. if wrong SSID specified then system continuously prints "no <SSID> found, reconnect after 1s". 
It really bothers me.

ourairquality, 2017-07-03:
I think this question has come up before. Some of the sdk calls to printf seem to use ets_printf, but the loader redirects them to printf. I think someone tried and might have been frustrated that gcc transforms printf into puts and putc sometimes and that broke their attempt to redirect them. Why don't we use a separate stub like etc_printf for all these to allow them to be redirected or quietened?

resetnow, 2017-07-03:
https://github.com/SuperHouse/esp-open-rtos/issues/188

xarrange, 2017-07-04:
Wow th works! Thanks!

 just added "printf ets_printf" to allsymbols.rename
and 
int ets_printf(const char *format, ...)
{
		return 0;
}

to my main.c

closed 2017-07-12 by UncleRus