SuperHouse/esp-open-rtos #402: DTLS - does it work?

malachib, 2017-07-04:
Has anyone had success with the mbedtls DTLS implementation?

I get so tantalizingly close, but can never quite get it working right.

projectgus, 2017-07-04:
IIRC the main sticking point is that it requires `recv(..., MSG_PEEK)` support in LWIP. Did you look into this?

You could either implement this in LWIP ("the right way", but probably tricky.) Or you could edit the net.c integration file for mbedTLS, and allow it to "peek" a UDP frame and save it somewhere there, to be returned the next time recv() is called.

malachib, 2017-07-04:
Excellent insight.  I did not know this was the case.  My own tests frequently fell apart at the end of the handshake phase inside a receive operation, and this could explain why.  This gives me some good direction, thank you.

Anyone else fiddle with this?

malachib, 2017-07-04:
@projectgus Do you recall if the issue you're describing is related to this output:

```
Waiting for a remote connection ... ok
  . Performing the DTLS handshake... failed
  ! mbedtls_ssl_handshake returned -0x7900
```

During the very tail end of an otherwise successful test using the standard dtls_server code from
https://github.com/ARMmbed/mbedtls/blob/development/programs/ssl/dtls_server.c

Adapted by using esp8266-specific & mbedtls_timing_set_delay/mbedtls_timing_get_delay

projectgus, 2017-07-05:
Don't know, sorry. I never tried DTLS, I just remember seeing it require MSG_PEEK support when I was doing the initial mbedTLS port. I thought I'd also documented that somewhere, but I guess I forgot (sorry.)

malachib, 2017-07-05:
@projectgus no worries, I will look into it over here and report my findings