SuperHouse/esp-open-rtos #406: Power usage is too high

Governa, 2017-07-04:
Comparing the power usage of nodemcu and esp-open-rtos, nodemcu uses about 50% less power with the same program. As both programs have an PWM enabled(100us wake up period), the difference seems to be related to the radio not entering in "modem sleep" mode correctly.

Does anyone have any ideas on how to solve this?

malachib, 2017-07-04:
FreeRTOS has an idle-hook feature, and ESP8266 has a minimal ambient-sleep mode.  I am unsure if esp-open-rtos out of the box enables those things, but that might be a place for you to start.

jones139, 2017-07-13:
@Governa Did you manage to solve this?  I have a problem with the device consuming about 80mA and I think getting it into modem-sleep mode should help a lot, but I am struggling to find a working example of its use.

jones139, 2017-07-15:
I have had a look at the Espressif RTOS Sdk, which has a wifi-station-machine-demo example, which seems to set MODEM_SLEEP mode using:
        bool s = wifi_set_opmode(mode);
        wifi_fpm_open();
        wifi_fpm_set_sleep_type(MODEM_SLEEP_T);
        wifi_fpm_do_sleep(0xFFFFFFFF);
The wifi_fpm_xxxx functions appear to be defined in libmain.a, with definitions in esp_wifi.h.   The definition of wifi_fpm_open() says you have to call that to enable the force sleep function as it is disabled by default.
Alas the wifi_fpm_xxx functions to not seem to be in the libmain.a file in esp-open-rtos.   Do you think I should be able to link against the version in the later Espressif SDK to get access to those functions?
Or have I missed something and esp-open-rtos already covers this?

Zaltora, 2017-07-18:
I didn't  found any function working to get moden sleep mode. It is like is not implemented in esp-open-rtos.

jones139, 2017-08-06:
Has anyone got any thoughts on how to go about fixing this - I'm happy to put some effort into a pull request for esp-open-rtos, because I don't like the Espressif RTOS sdk (all the GPIO stuff is so complicated!)....but I'm not really sure where to start - there must be some registers that need to be written to the chip to put it into MODEM_SLEEP mode to allow us to create an esp-open-rtos version of wifi_fpm_set_sleep_type etc    I'll ask Espressif, but not sure how helpful they will be - any idea what the other options are for fixing this?   Othewise I'll have to give up on esp-open-rtos and learn the Espressif sdk instead....

resetnow, 2017-08-06:
>not sure how helpful they will be

Worth a try, but this will most likely yield nothing

>there must be some registers that need to be written

That's about all we know :) 

Debugging and implementing sleep mode will probably require lots of reverse engineering. If you'd like to try that, just say — there are plenty of tools available and I will happily provide some links. For a first step, I think one needs to check whether Espressif SDK uses tick suppression feature of FreeRTOS (portSUPPRESS_TICKS_AND_SLEEP) and what how it's implemented. For wifi_fpm_* stuff again, reverse engineering and decompilation may be the only way.

jones139, 2017-08-06:
Thanks - I'd interested in any pointers you have into reverse engineering it - I'm not sure how far I'll get though, because my projects are very limited by spare time availability, and I'm not even sure the ESP8266 will be the final board I use - I'd really prefer a Bluetooth LE equivalent rather than wifi for comms, so I may give up if it sounds too hard - so don't put too much effort into digging things out for me! 
I was wondering about  a simpler way out which may be to link to the current version of the Espressif SDK that includes the functions I need, to essentially use a hybrid between the Espressiff SDK and esp-open-rtos - I had a quick go a couple of weeks ago and got loads of linker errors, but I am a bit out of practice so may have made a silly mistake - do you think that approach would be 1) feasible and 2) legal from a license perspective?

resetnow, 2017-08-06:
>reverse engineering

* mailing list: https://groups.google.com/forum/#!forum/esp8266-re
* tooling: radare2, IDA xtensa plugin
* wiki: http://esp8266-re.foogod.com/ (it's down, but should be available in cache/archives)

Also, @ourairquality did a lot of reverse engineering and could probably recommend an optimal approach.

>feasible

I don't think they keep binary compatibility, especially for internal functions. You may try to extract separate object files and objects, though.

ourairquality, 2017-08-07:
Managed to get the wificfg example working in part using recent sdk libraries, and with the current FreeRTOS and lwip. There were are lot of hacks to get there. Some patches have been submitted to help bring this closer, such as adding a argument to interrupt handlers, moving the sysparams sectors etc.

I doubt it would be possible to split it up much more, not possible to pick just one of the libraries. For example, there seems to have been a significant change to how the wdev nmi is handled, and that seems baked into the binaries in many places. It was a lot of work just to separate it from the bundled version of lwip. Not sure if or when I'll look at that hack again, and I have not looked at the power usage either but it would be nice to have some control over it.

There appears to be a chunk of code in libpp, pm.o, and it's not clear if it is actually used.

What happens if the network mode is changed to 'none', does it turn off the radios and lower power consumption? I could live with just waking up the radio periodically and then turning it off again if that would have a similar effect.

jones139, 2017-08-07:
@ourairquality - Thanks -  I'll have a go at doing sdk_wifi_set_opmode(NULL_MODE); which I think is what you mean?  Unfortunately I don't have a proper meter with me, just a USB tester that measures in 10mA steps, but it looks very different to what I was getting with a DVM so I don't think I believe the results yet...