SuperHouse/esp-open-rtos #409: Not able to assign static IP to the ESP in Station Mode?

rameshKrSah, 2017-07-06:
I am trying to assign static IP to the ESP when it tries to connect to my WiFi network. I have used the code given below: 
 
`void _is_esp_connected_to_wifi(void){
	int status = sdk_wifi_station_get_connect_status();

	while(status != STATION_GOT_IP){
		vTaskDelay(100);
		status = sdk_wifi_station_get_connect_status();
//		TRACE("Connection Status: %d -- %s", status, sdk_wifi_status_2_string(status));
	}

	return;
}

void _station_init(void){

	char ssid[] = "WCS";
	char password[] = "RTSR&DWCS";
	struct sdk_station_config _station_info;
	struct ip_info static_ip_info = {0};

	TRACE("SSID:%s, Password: %s", ssid, password);
	strcpy((char *)_station_info.ssid, ssid);
	strcpy((char *)_station_info.password, password);
	_station_info.bssid_set = 0;

	//Delete all existing wifi stuff in the EEPROM
	sdk_wifi_station_disconnect();
	sdk_wifi_set_opmode(NULL_MODE);
	dhcpserver_stop();
	vTaskDelay(500);

	IP4_ADDR(&static_ip_info.ip, 192, 168, 5 ,55);
	IP4_ADDR(&static_ip_info.gw, 192, 168, 5, 1);
	IP4_ADDR(&static_ip_info.netmask, 255, 255, 255, 0);
	TRACE("static ip set status : %d", sdk_wifi_set_ip_info(STATION_IF, &static_ip_info));
	vTaskDelay(50);

//Must call sdk_wifi_set_opmode before sdk_wifi_station_set_config
	sdk_wifi_set_opmode(STATION_MODE);
	sdk_wifi_station_set_config(&_station_info);
	sdk_wifi_station_connect();
	_is_esp_connected_to_wifi();
	TRACE("ESP Connected to WiFi %s", ssid);

}

void station_connect(void *pc){
	struct ip_info get_static_ip = {0};

	_station_init();
	sdk_wifi_get_ip_info(0, &get_static_ip);
	TRACE("Obtained IP Address: %d.%d.%d.%d",ip4_addr1(&get_static_ip.ip),
			ip4_addr2(&get_static_ip.ip),ip4_addr3(&get_static_ip.ip),ip4_addr4(&get_static_ip.ip));
	while(1){
		vTaskDelay(1000);
		TRACE("static ip test ");
	}

	vTaskDelete(NULL);
}

void user_init(){
//	sdk_system_update_cpu_freq(SYS_CPU_160MHZ);

#if	1
	//sdk_wifi_set_opmode(NULL_MODE);
	//sdk_wifi_station_set_auto_connect(0);
#endif

#if ENABLE_DEBUG == 1
	uart_set_baud(0, 74880);
	iomux_set_function(gpio_to_iomux(2), IOMUX_GPIO2_FUNC_UART1_TXD);
#endif

	TRACE("SDK Version: %s", sdk_system_get_sdk_version());
	TRACE("Free Heap Size: %d", sdk_system_get_free_heap_size());
	TRACE("Chip ID: %d", sdk_system_get_chip_id());
	TRACE("user_init running for static ip test ....");

	xTaskCreate(station_connect, "station", 500, NULL,14, NULL);
}`

Also the dhcp server keep on running, although i have stopped it in the above code .

Is this the right way of doing this and if not can someone point it out what am I doing wrong. Thanks.

ourairquality, 2017-07-06:
I don't think a static IP address ever worked for a station, but an attempt to fix it has been made in the rewrite for lwip v2 in https://github.com/SuperHouse/esp-open-rtos/pull/389

Much of the logic seems to be in open_esplibs/libwpa/wpa_main.c:sdk_eagle_auth_done() see the trailing code in that function, so if you are still seeing problems then perhaps you could trace the logic there. The start/stop functions are in libmain/user_interface and might need some work too.

rameshKrSah, 2017-07-06:
@ourairquality thanks for your reply, but after an hour of more fiddling, i have successfully assigned static IP to the esp. I have checked the program for different IPs and it worked all fine. Here is he code fragment:
`
#include "espressif/esp_common.h"
#include "esp/iomux.h"
#include "esp/uart.h"
#include <string.h>
#include "FreeRTOS.h"
#include "task.h"
#include "semphr.h"
#include "uart0.h"
#include "open_esplibs.h"
#include <lwip/sockets.h>
#include "esplibs/libmain.h"
#include "esplibs/libnet80211.h"
#include "esplibs/libpp.h"
#include "esplibs/libwpa.h"
#include "espressif/user_interface.h"
#include <lwip/dhcp.h>

void _station_init(void){

	char ssid[] = "WCS";
	char password[] = "RTSR&DWCS";
	struct sdk_station_config _station_info;
	struct ip_info static_ip_info = {0};

	TRACE("SSID:%s, Password: %s", ssid, password);
	strcpy((char *)_station_info.ssid, ssid);
	strcpy((char *)_station_info.password, password);
	_station_info.bssid_set = 0;

	//Delete all existing wifi stuff in the EEPROM
//	sdk_wifi_station_disconnect();
//	sdk_wifi_softap_stop();
//	sdk_wifi_station_stop();
//	sdk_wifi_set_opmode(NULL_MODE);
//	vTaskDelay(500);

	TRACE("dhcp status : %d", sdk_wifi_station_dhcpc_status());
	IP4_ADDR(&static_ip_info.ip, 192, 168, 5 ,133);
	IP4_ADDR(&static_ip_info.gw, 192, 168, 5, 1);
	IP4_ADDR(&static_ip_info.netmask, 255, 255, 255, 0);
	sdk_wifi_station_dhcpc_stop();
	vTaskDelay(50);

//Must call sdk_wifi_set_opmode before sdk_wifi_station_set_config
	sdk_wifi_set_opmode(STATION_MODE);
	TRACE("static ip set status : %d", sdk_wifi_set_ip_info(STATION_IF, &static_ip_info));
	sdk_wifi_station_set_config(&_station_info);
//	sdk_wifi_station_start();
	sdk_wifi_station_connect();
	_is_esp_connected_to_wifi();
	TRACE("ESP Connected to WiFi %s", ssid);
}
`

closed 2017-07-06 by rameshKrSah