SuperHouse/esp-open-rtos #427: [Feature request] Integration of esp-gdbstub possible?

flannelhead, 2017-07-30:
There's a [GDB stub](https://github.com/espressif/esp-gdbstub) for ESP8266 made by Espressif. The software running on the ESP8266 can be debugged at runtime with the help of the stub.

Do you think it would be possible / feasible to integrate this stub into this project? Mainly, it would be nice to have as a part of the build system and automatically linked into debug builds.

I'm probably going to try that out in the near future. If I manage to do it well, I could send a PR.

resetnow, 2017-07-30:
https://github.com/resetnow/esp-gdbstub

It's not in the main repo because of its license.

flannelhead, 2017-07-31:
Thanks for the reminder, I somehow managed to overlook the licensing issue.

closed 2017-07-31 by flannelhead