SuperHouse/esp-open-rtos #429: i2c + MMA7455L - Input/Output ERROR

AdahilMuniz, 2017-08-01:
Hi, I have a problem related to the i2c. My application needs to take acceleration from a MMA7455L. So, first, I need to configure the sensor writing in the register 0x16. But i'm receiving a I/O error. Can someone help me?

Module: NodeMCU V1.0 || ESP-12E
Sensor: MMA7455L

Code:

`#include "i2c/i2c.h"

#define SCL (0)
#define SDA (2)

void configure();

void user_init()
{
    gpio_enable(BT, GPIO_INPUT);
    uart_set_baud(0, 115200);
    printf("SDK version:%s\n\n", sdk_system_get_sdk_version());
    i2c_init(SCL, SDA);

    configure();
}

void configure(){

    uint8_t addr = 0x1D;
    const uint8_t reg = 0x16;
    const uint8_t cmd = 0x01;

    int error = i2c_slave_write(addr, &reg, &cmd, 1);
    if(error!=0){
        printf("ERROR\n");
        if(error == -EBUSY){
            printf("Device Busy!\n");
        }
        if(error == -EIO){
            printf("Input/Output Error!\n");
        }
    }

}`

andrewclink, 2017-08-01:
Unfortunately, I've had some sketchy behavior with I2c. Double and triple check your connections and pull-ups, and consider modifying the i2c bit delay to slow down communications.

A logic analyzer (or DSO or maybe even analog scope) would go a long ways toward debugging this.

resetnow, 2017-08-01:
I would also suggest changing pullups to 4.7kOhm and maybe trying a lower frequency (if you are using 400k)

AdahilMuniz, 2017-08-02:
@andrewclink, I used another code(Arduino) with the same connections that I have now and the applicantion worked. So, I think that the connections are right . I will try to change the delay, but how can I do this? Thank you for your help.

AdahilMuniz, 2017-08-02:
@resetnow how can I change the pullups and the frequency? Thank you for help!

resetnow, 2017-08-02:
>how can I change the pullups 

If you haven't already, you need to physically connect resistors to I2C lines (if your board doesn't have them already).

>how can I change the pullups and the frequency

Currently there is a line of code that ended up in the repo by accident. You can remove `#define I2C_FREQUENCY_400K` from i2c.h

AdahilMuniz, 2017-08-03:
@resetnow I tried to remove the `I2C_FREQUENCY_400K`, but nothing happened. However, I know that the frequency of the device is 100K.
I will try to connect the resistors, but as i said before, with another code(Arduino) the configuration works.
Anyway, thank you very much for your help.