SuperHouse/esp-open-rtos #442: Example spiffs - Error mount SPIFFS

fvpalha, 2017-09-08:
Hi.

My ESP8266 models are ESP-12 and ESP-12e.

**SPIFFS_SINGLETON = 1**

```
pp_task_hdl : 3fff1ec0, prio:14, stack:512
pm_task_hdl : 3fff2350, prio:1, stack:176
frc2_timer_task_hdl:3fff4f18, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Sep  8 2017 08:45:09

phy ver: 273, pp ver: 8.3

SDK version:0.9.9

mode : softAP(5e:cf:7f:05:bc:bd)

add if1

bcn 100

esp_spiffs_init()

SPIFFSbuf_size=512, fds_buf_size=220, example_write_file - Error opening file

example_read_file_posix - Error opening file

example_read_file_spiffs - Error opening file

Total: 0 bytes, used: 1835104325 bytes.
```

**SPIFFS_SINGLETON = 0**

```
pp_task_hdl : 3fff1ee8, prio:14, stack:512
pm_task_hdl : 3fff2378, prio:1, stack:176
frc2_timer_task_hdl:3fff4f40, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Sep  8 2017 08:52:38

phy ver: 273, pp ver: 8.3

SDK version:0.9.9

mode : softAP(5e:cf:7f:05:bc:bd)

add if1

bcn 100

esp_spiffs_init(0x200000, 0x10000)

SPIFFS memory, work_buf_size=512, fds_buf_size=220, cache_buf_size=1400

Error spiffs mount: -10050

test_task - Error mount SPIFFS

example_write_file - Error opening file

example_read_file_posix - Error opening file

example_read_file_spiffs - Error opening file

Total: 0 bytes, used: 1835104325 bytes.
```

How I can solve this?

sheinz, 2017-09-09:
Hi @fvpalha 

Judging from the source code  `SPIFFS_ERR_INTERNAL             -10050` is caused mainly due to wrong flash size.
The example expects 4MB flash chip. 
You can determine you flash chip size with:
```
> esptool.py flash_id
esptool.py v1.2
Connecting...
Manufacturer: ef
Device: 4016
```
You can search flash chip id on the Internet.

You can also try to move the spiffs region to the first 1MB:
```diff
--- a/examples/spiffs/Makefile
+++ b/examples/spiffs/Makefile
@@ -3,8 +3,8 @@ EXTRA_COMPONENTS = extras/spiffs
 FLASH_SIZE = 32

 # spiffs configuration
-SPIFFS_BASE_ADDR = 0x200000
-SPIFFS_SIZE = 0x010000
+SPIFFS_BASE_ADDR = 0xc0000
+SPIFFS_SIZE = 0x10000

 # SPIFFS_SINGLETON = 0  # for run-time configuration
```

fvpalha, 2017-09-13:
Hi @sheinz 

Thank you.

```
$ esptool.py flash_id
esptool.py v1.2
Connecting...
Manufacturer: e0
Device: 4016
```
I changed the Makefile as was suggested.

```
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0060
Compressed 3104 bytes to 2169...
Wrote 3104 bytes (2169 compressed) at 0x00000000 in 1.2 seconds (effective 21.2 kbit/s)...
Hash of data verified.
Compressed 2048 bytes to 23...
Wrote 2048 bytes (23 compressed) at 0x00001000 in 0.0 seconds (effective 453.5 kbit/s)...
Hash of data verified.
Compressed 250800 bytes to 184924...
Wrote 250800 bytes (184924 compressed) at 0x00002000 in 97.3 seconds (effective 20.6 kbit/s)...
Hash of data verified.
Compressed 65536 bytes to 228...
Wrote 65536 bytes (228 compressed) at 0x000c0000 in 0.1 seconds (effective 3646.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
gtkterm &

```

The result:

```
pp_task_hdl : 3fff1ee8, prio:14, stack:512
pm_task_hdl : 3fff2378, prio:1, stack:176
frc2_timer_task_hdl:3fff4f40, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Sep 13 2017 16:43:15
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : softAP(5e:cf:7f:05:bc:bd)
add if1
bcn 100
esp_spiffs_init(0x200000, 0x10000)
SPIFFS memory, work_buf_size=512, fds_buf_size=220, cache_buf_size=1400
Error spiffs mount: -10050
test_task - Error mount SPIFFS
example_write_file - Error opening file
example_read_file_posix - Error opening file
example_read_file_spiffs - Error opening file
Total: 0 bytes, used: 1835104325 bytes.
```

Thank you again.

chenli2015, 2018-03-19:
I also have this problem, and tried all ways but failed! My suggestion is that don't touch spiffs now~~~~

fvpalha, 2018-03-19:
Hi @chenli2015 
I stopped all development with spiffs until the moment.
I am waiting for more stablity and user friendly.