SuperHouse/esp-open-rtos #445: read(...) and write(...) no longer works with LWIP sockets

kanflo, 2017-09-18:
I noticed read(...) and write(...) no longer can operate on LWIP sockets. Does anyone know if is this intended or accidental? It used to work about a year ago with the following snippet working as a web server. Today write(...) sends the HTTP response to stdout (the uart) and read(...) returns nothing (probably as the part is idle).

```
struct sockaddr_in server_addr, client_addr;
int server_sock, client_sock;
socklen_t sin_size;
bzero(&server_addr, sizeof(struct sockaddr_in));
server_addr.sin_family = AF_INET;
server_addr.sin_addr.s_addr = INADDR_ANY;
server_addr.sin_port = htons(80);

server_sock = socket(AF_INET, SOCK_STREAM, 0);
bind(server_sock, (struct sockaddr *)(&server_addr), sizeof(struct sockaddr));
listen(server_sock, 5);
sin_size = sizeof(client_addr);
for (;;) {
    if ((client_sock = accept(server_sock, (struct sockaddr *) &client_addr, &sin_size)) < 0) {
        printf("accept fail\n");
        continue;
    }
    printf("Client from %s:%d\n", inet_ntoa(client_addr.sin_addr), htons(client_addr.sin_port));
    int opt = 50;
    lwip_setsockopt(client_sock, SOL_SOCKET, SO_RCVTIMEO, &opt, sizeof(int));
    while ((recbytes = read(client_sock, buffer, BUF_SIZE)) > 0) {
        printf("%c", recbytes);
        buffer[recbytes] = 0;
    }
    sprintf(buffer, "HTTP/1.1 200 OK\nContent-Type: image/jpeg; charset=utf-8\nConnection: close\n\n");
    write(client_sock, buffer, strlen(buffer));
    close(client_sock);
}
```

ourairquality, 2017-09-18:
I have not seen this problem with the lwip v2 branch. There SO_RCVTIMEO takes a `struct timeval`, and might that be a problem in your code. For some example code see the wificfg web server.

kanflo, 2017-09-20:
Thanks, I found this when sorting out some code rot needed for me to test your PR but I will skip digging into this and go to testing. Also, thanks for pointing out the mistake regarding SO_RCVTIMEO.

kanflo, 2017-10-07:
Please see #454. I will move the ```read```discussion to this issue. I found a workaround by setting LWIP_POSIX_SOCKETS_IO_NAMES to 0 in a local lwipopts.h:

```
#include <../lwip/include/lwipopts.h> // Hackishly include EOR lwipopts.h :-/
#define LWIP_POSIX_SOCKETS_IO_NAMES 0
```

Still, I need to look into why lwip does not call _read_stdin_r...

ourairquality, 2017-10-07:
Thank you. It's amazing this even worked, I had not seen this problem. Changing `LWIP_POSIX_SOCKETS_IO_NAMES` to 0 looks like the correct path. We need to also coming up with a strategy for allocation the descriptor index space so that the dispatching in core/newlib_syscalls.c can work.

The number of file descriptors used by lwip appears to be derived from MEMP_NUM_NETCONN so perhaps we could use that so set the block allocated to lwip. We could set the LWIP_SOCKET_OFFSET to the start of the block allocated to lwip so that the same fd could be used with `read` and `lwip_read` etc.

spiffs_fd_find_new appears to allocate from zero too. That also looks broken, e.g. what if it allocated a descriptor that matches one of stdin or stdout, then it would not work. The size of the spiffs fd allocation seems to be dynamic, so perhaps we could place that last.

Or perhaps we just allocate good blocks for each of these, the descriptor is a 32 bit integer so there is lots of space, so perhaps 4096 descriptors each for the newlib, then lwip, then spiffs.

It also looks like we need to do something with readv and writev and and open and close at least.

What do people think?

ourairquality, 2017-10-08:
Unfortunately lwip does not build with LWIP_SOCKET_OFFSET not zero and FD_SET defined externally as we do in newlib. Obviously `select` is not going to work with descriptors across different drivers, but perhaps lwip sockets could do better here, so that it works with a non-zero LWIP_SOCKET_OFFSET - shall have to explore this lwip limitation first.

We probably also want to keep the FD_SET_SIZE small too, and the default for newlib is 64, so we would not want to allocate large blocks of index space for each driver as suggested above, rather pack them in. I think newlib needs only 0 to 2 for the std streams, and the default number of sockets for lwip is only 4.

kanflo, 2017-10-10:
Now that you mention it I recall LWIP_SOCKET_OFFSET from an earlier discussion (earlier as a year or two ago). Seems odd LWIP won't build with that feature. Is it a problem with our integration or on their master? I would be great to be able to use read/write on any FD but if not we need to make sure the behaviour is consistent. I was quite surprised the first time my image capture program sent a JPEG on the UART ;)

ourairquality, 2017-10-11:
I am not sure what the lwip issue is with LWIP_SOCKET_OFFSET. A quick inspection, of socket.c, found only one issue. I have not actually run any tests of `select()` though. File a bug asking for feedback see https://savannah.nongnu.org/bugs/index.php?52190

Perhaps it was just that `select()` will not work across descriptors of different drivers that worried them. So it can not wait on both uart i/o and/or lwip socket i/o and/or spiffs i/o, rather just on the lwip descriptors. That might be a show stopper for some implementations, but is a limitation we can work around in practice in our limited apps.