SuperHouse/esp-open-rtos #454: examples/sysparam_editor causes WD to bite

kanflo, 2017-10-07:
Following the great work on #437 I noticed the sysparam editor example causes the watchdog to bite. Currently digging into this.

kanflo, 2017-10-07:
This seems to be related to the ```read(...)``` stuff discussed in #437 and also in #445. A call to ```read(...)``` may end up in newlib or in lwip. Eg. ```examples/terminal/``` ends up in ```_read_stdin_r```but another program I have ends up in ```lwip_read``` for this code: 

```
if (read(0, (void*)&ch, 1)) { // 0 is stdin
```

I for one would prefer knowing where a call to ```read``` ends up ;) I guess the old lwip could differ between stdin and lwip sockets.

ourairquality, 2017-10-07:
I had not seen any watchdog timeouts working on this code, and I use the sysparam editor occasionally. The `read` function does need to dispatch to different code depending on the descriptor. In this example it should no be calling into lwip, but will call into lwip if the descriptor is a socket. The rtos is preemptive, so even if this spins it should not cause a watchdog timeout.

I think it would be useful to check some of the read/write paths from newlib to the drivers, but it is not obvious how that could be causing a watchdog timeout.

Perhaps it is just that the code is waiting some time at `read` so that it is prone to hitting some other issue while doing so?

kanflo, 2017-10-07:
I will move the read path discussion to #445 as it is a separate issue (methinks). I will prioritise #445 as it is currently a blocker for me.

PR #455 (2017-10-10): Added fflush for examples/terminal echo
As the terminal example was meant to echo characters typed I added fflush since the line buffering policy has changed.

While on the topic, should we make IRQ driven stdin standard (which would fix #454)?
closed 2017-10-10 by kanflo