SuperHouse/esp-open-rtos #456: Adding libesphttpd

nochkin, 2017-10-08:
Hey folks.

I'm so in love with esp-open-rtos and [libesphttpd](https://github.com/Spritetm/libesphttpd), so I think these two must live together.
For that purpose I recently modified libesphttpd to work with esp-open-rtos (as one of the "extras").

Things currently work:
- HTTP web server
- Static files using heatshrink and gzip compression
- CGI dynamic content
- WebSocket
- All built-in tests passed

"Bonus" features:
- WiFi configuration using web interface
- OTA update over http using a simple **curl** command (implemented using rboot-ota)

I think it should be fair for me to share my work of porting the libesphttpd.
It's currently implemented as a fork of libesphttpd. Is it acceptable way to contribute or it should be integrated into the original project first?
I'm kinda new for this stuff. Please let me know what do you think.

ourairquality, 2017-10-08:
It seems suitable for the extras and an example would help too. Many of the extras reference external repos, and if you plan to maintain and support it then you could do that.

If the changes are specific to esp-open-rtos then some conditional code seems fine, otherwise I think it helps the communities to make an effort to get changes landed upstream.

I suggest keeping a repo that tracks the upstream, with a branch for the esp-open-rtos changes, and then try to whittle down the changes in time. This will also make it a lot easier to update the code to track upstream changes.

ourairquality, 2017-10-08:
Another wip: https://github.com/Spritetm/libesphttpd/pull/38

nochkin, 2017-10-08:
I have no problems maintaining my own fork just because I don't have a way to make a full 100% test to make sure the upstream works with other projects it may be linked to.
And yes, I have a working example too.

Maybe in the future, I'll make the things into upstream so the fork will not longer be needed.
Thank you for your feedback, I'll start working on it and propose it as a pull request.

nochkin, 2017-10-09:
I created the initial pull request: https://github.com/SuperHouse/esp-open-rtos/pull/458
That's currently done as a fork from the original libesphttpd project.

closed 2017-10-24 by nochkin