SuperHouse/esp-open-rtos #467: sdk API

slowkdj, 2017-10-13:
I started using this SDK. Where can I find the API

kezl, 2017-10-16:
I don't think there is any official documentation as such, but it is fairly easy to get clued up by reading through the examples and also the header files in the extras directory. For the rtos side of things the official FreeRTOS documentation is really helpful and is available as free pre-release copies, see: [http://www.freertos.org/Documentation/RTOS_book.html](url)

Bear in mind that EOR itself is based on an earlier version of the official sdk so a lot of the (particularly older) examples on the official forums may work with little or no tweaking.

kanflo, 2017-10-18:
Hi @slowkdj. Please have a look at the ```examples/``` folder which demonstrates all the different features of ESP Open RTOS and the different additions and drivers people have contributed over the years. These can be found in ```extras/```.

closed 2017-10-18 by kanflo