SuperHouse/esp-open-rtos #472: examples/sysparam_editor immediately crashes

jeffsf, 2017-10-18:
The sysparam_editor example does not run much past the initial header printed, constantly crashing.
Adafruit Huzzah, Adafrtuit FTDI Friend. build system on Mac OS X El Cap, esp-open-sdk toolchain, known to build and run esp-open-rtos software.

```
esp-open-rtos/examples/sysparam_editor]$ make test -j4 ESPPORT=/dev/tty.usbserial-AH02F0VW
[...]
Wrote 253952 bytes at 0x2000 in 22.0 seconds (92.2 kbit/s)...
Leaving...
../..//utils/filteroutput.py --port /dev/tty.usbserial-AH02F0VW --baud 115200 --elf ./build/sysparam_editor.out
Opening /dev/tty.usbserial-AH02F0VW at 115200bps...

ets Jan  8 2013,rst cause:2, boot mode:(1,7)

ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

press Reset switch, after serial-line "garbage" clears

```
ESP-Open-SDK ver: 0.0.1 compiled @ Oct 17 2017 20:22:24
phy ver: 273, pp ver: 8.3

mode : null

Welcome to the system parameter editor!  Enter 'help' for more information.

[current sysparam region is at 0x001f7000 (4 sectors)]

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16

0x40100000: _stext at ??:?

tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   16 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3fff1de8, prio:14, stack:512
pm_task_hdl : 3fff2368, prio:1, stack:176
frc2_timer_task_hdl:0x3fff57d8, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Oct 17 2017 20:22:24
phy ver: 273, pp ver: 8.3
```

[repeats, _ad infinitum]_

commit 87a3503f931021f28d4dbe083f5dec08c626354b
Author: Ruslan V. Uss <UncleRus@users.noreply.github.com>
Date:   Thu Oct 12 17:42:34 2017 +0500

    I2C code formatting, README fix (#462)

kanflo, 2017-10-18:
I did spot something like this in #454. Could you try and replace the line reading part with the one used in ```examples/terminal```?. You will need to pull ```examples/terminal/FreeRTOSConfig.h``` also.

Magnietto, 2017-11-01:
Yeah using an adapted version of read instead of getchar worked for me.

ourairquality, 2017-12-08:
Was able to reproduce this, and the problem appears to be that `getchar()` acquires a lock while waiting, and this blocks the wdt paths, probably a call to malloc. This should not be a problem, except that we enter a critical region for any newlib lock. The proper fix will be to implement the newlib locking functions in terms of FreeRTOS mutexes. A simpler workaround is to replace `getchar()` with `getchar_unlocked()`.