SuperHouse/esp-open-rtos #477: LWIP: lwip_setsockopt for SO_RECVTIMEO and SO_SENDTIMEO unimplemented

jeffsf, 2017-10-26:
After beating on why blocking socket reads weren't working, enabling LWIP_DEBUG and SOCKETS_DEBUG revealed that it is apparently not implemented. In the following trace output, the first column is seconds (from task ticks). As can be seen, the LWIP debug indicates that SO_RECVTIMEO is not implemented. The over 13-second delay is also inconsistent with setting the timeout with `struct timeval so_timeout = { .tv_sec = 0, .tv_usec = 1000000 }` (1 second).

This makes blocking read with `lwip_recv()` non-functional.
```
SO_RECVTIMEO: 0x1006
lwip_setsockopt(3, IPPROTO_IP, UNIMPL: optname=0x1006, ..)
re: -1
5.400 FreeRTOS_read(): Read timeout: 0.0000
5.410 FreeRTOS_read(): FreeRTOS_read() calling _recv()
lwip_recvfrom(3, 0x3fff8c30, 1, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0
10.010 beat_task(): beat...
lwip_recv_tcp: netconn_recv err=-15, pbuf=0
lwip_recv_tcp: p == NULL, error is "Connection closed."!
lwip_recvfrom(3):  addr=37.187.106.16 port=1883 len=0
19.110 FreeRTOS_read(): FreeRTOS_read() exit of _recv(), rc: 0
lwip_getsockopt(3, SOL_SOCKET, SO_ERROR) = 0
19.120 FreeRTOS_read(): No socket error reported
```

_Note: The FreeRTOS_read() function is so named to replace the function within the upstream paho.mqtt.embedded-c source. The internals are implemented with lwip_*_

ourairquality, 2017-10-26:
```
   const struct timeval timeout = { 1, 0 }; /* 1 second timeout */
   setsockopt(s, SOL_SOCKET, SO_RCVTIMEO, &timeout, sizeof(timeout));
   setsockopt(s, SOL_SOCKET, SO_SNDTIMEO, &timeout, sizeof(timeout));
```

jeffsf, 2017-10-26:
_**Not a bug**_

Found _my_ error in the twisted set of header defines needed to adapt the MQTTFreeRTOS.c code from paho.mqtt.embedded-c to LWIP.

For anyone trying to debug LWIP, changing `lwipopts.h` to include
```
#define LWIP_DEBUG
#define SOCKETS_DEBUG                   LWIP_DBG_ON
```
provides detailed tracing of the LWIP internals. Other categories available as well, some with levels of detail that can be set.

closed 2017-10-26 by jeffsf