SuperHouse/esp-open-rtos #480: i2c clock timing

gschorcht, 2017-10-31:
When I was investigating the clock stretching problem in #471 with a logic analyzer, I realized also that the clock frequency produced on SCL is varying a lot for write operations. It is much too high for logic 0. For example, at ```I2C_FREQ_100K``` the clock frequency is about 88 kHz for logic 1, but almost 160 kHz for logic 0.

The reason is, that in read operations only ```read_sda``` is used while in write operations ```read_sda``` is used for logic 0 and ```clear_sda``` for logic 1. ```read_sda``` includes an additional delay to avoid arbitration errors during the stop condition.

That is, it seems that the delay produced by ```i2c_delay``` is too small. The frequency is appropriate for read operations and logic 0 in write operations only because of the additonal delay in ```read_sda```.

I have changed my ```i2c.c``` in that way, that I removed the additional ```i2c_delay``` from ```read_sda``` and placed an additional ```read_sda``` and ```i2c_delay``` in stop condition before ```read_sda``` is used to  check SDA for high level. Furthermore, I increased the delay values for I2C_FREQ_100K and I2C_FREQ_80K at 80 MHz system clock. Unfortunatly, I don't know how to do that for 160 MHz system clock.

Now, the SCL clock seems to be quite stable with following frequencies:

```
I2C_FREQ_80K    mean 78 kHz, varies from 75 kHz to 80 kHz
I2C_FREQ_100K   mean 95 kHz, varies from 92 kHz to 100 kHz
```

If you want, I could provide my changes by a pull request.

gschorcht, 2017-11-28:
Here are some snapshots that show the timing for write operation.

Logical 0:
[image]

Logical 1:
[image]

UncleRus, 2017-12-09:
Fixed by #503

closed 2017-12-09 by UncleRus