SuperHouse/esp-open-rtos #482: basename() declared in <libgen.h> but not resolved by linker

jeffsf, 2017-11-09:
With `#include <libgen.h>` references to usages of `basename()` compile, but result in 
```
undefined reference to `basename'
```
during link.

Inability to resolve link also found with `<string.h>`

`esp-open-rtos/libc/xtensa-lx106-elf/include/libgen.h`

jeffsf, 2017-11-10:
Public-domain code (at least as indicated in the source file) for basename is available at /usr/src/contrib/gcclibs/libiberty/basename.c on a FreeBSD 11.1-RELEASE distribution.

https://svnweb.freebsd.org/base/releng/11.1/contrib/gcclibs/libiberty/basename.c?revision=320486&view=markup

Comment out "unneeded" #include statements and it compiles and appears to run under espo-open-rtos
```
/* #ifdef HAVE_CONFIG_H */
/* #include "config.h" */
/* #endif */
/* #include "ansidecl.h" */
/* #include "libiberty.h" */
/* #include "safe-ctype.h" */
```

ourairquality, 2017-11-10:
Not sure what is going wrong here, but `basename` appears to be defined in newlib at newlib/libc/string/gnu_basename.c and newlib/libc/unix/basename.c

closed 2017-12-02 by UncleRus

jeffsf, 2018-02-05:
I've got a fresh pull of esp-open-rtos as well as a fresh build of esp-open-sdk and not only does `basename` not resolve in the link, but also now having similar problems with `settimeofday`

Will get some minimal examples

jeffsf, 2018-02-20:
Resolved by https://github.com/SuperHouse/esp-open-rtos/pull/571