SuperHouse/esp-open-rtos #487: wifi and uart problem

Zaltora, 2017-11-20:
Can someone can help me to found how disable this information on uart0 when i use wifi:
```
add 1
aid 1
station: xx:xx:xx:xx:xx:xx join, AID = 1
station: xx:xx:xx:xx:xx:xx leave, AID = 1
rm match
```
i use UART and these informations broke my com.
i use `uart_putc(0,c);` . If i cant disable it, maybe disable/replace printf can be a "FIX", how do it ?

kanflo, 2017-11-21:
Please refer to #400

Zaltora, 2017-11-21:
I apply the solution at the end of #400  with an additional change to work.
-Add to your project:
```
int ets_printf(const char *format, ...)
{
return 0;
}
```
- add  `printf ets_printf` to "lib/allsymbols.rename"
- comment `ets_printf = printf;` at "ld/programm.ld"

ourairquality, 2017-11-22:
Could this be done by adding a dummy function for stdout output see `set_write_stdout()` in newlib_syscalls.c. I had been thinking of trying to redirect this all to a small in memory log as the uart messages are not accessible on a remote device anyway, that way it could be viewed via a web interface etc.