SuperHouse/esp-open-rtos #488: Is there a way to disable printf() buffering?

funnydog, 2017-11-21:
Hi,

A while ago I wrote an application working on esp-open-rtos that used printf() to simulate a simple shell and it worked well.

A few days ago I decided to update esp-open-rtos to HEAD and after fixing some fallouts, mainly the uart ISR definition and the interrupt registration, I noticed that the prompt isn't printed on time, but only after the user input: it seems printf() is now buffered.

I had to add a fflush() after the printf("$ ").

Sample of the code:

```
	while (!end) {
		/* command prompt */
		printf("$ ");
		fflush(stdout);

		read_line(line, sizeof(line));
		split_line(args, 10, line);
		execute_cmd(conf, args);
	}
```
Since I don't want to waste memory I tried to disable the buffering using the standard way: setbuf(stdout, NULL);

Unfortunately it didn't work. Is there a way to disable the buffering and return to the old behaviour?

Thank you.

ourairquality, 2017-11-22:
The newlib code writes "You can also use NULL as the value of buf, to signal that the setbuf function is to allocate the buffer." so perhaps try `setvbuf()`. The initialization code is in `findfd.c:stdout_init()` if you wished to compile up a modified version of newlib.

funnydog, 2017-11-22:
OK thank you I'll look at the code and figure out how to do it.

closed 2017-11-22 by funnydog