SuperHouse/esp-open-rtos #489: commit ebdd2f9 broke SPIFFS POSIX API

funnydog, 2017-11-21:
Hi,

I noticed in one of my projects that the commit ebdd2f9 broke the SPIFFS POSIX API.
If I revert it the POSIX API works again.

To rule out any error on my part I also verified that the SPIFFS example is also affected.

Here follows the complete git bisect log:

```
git bisect start
# bad: [d36e9d65a051ac0cee7253e43b5141b14fc7cddf] Fix a printf issue. (#474)
git bisect bad d36e9d65a051ac0cee7253e43b5141b14fc7cddf
# bad: [d36e9d65a051ac0cee7253e43b5141b14fc7cddf] Fix a printf issue. (#474)
git bisect bad d36e9d65a051ac0cee7253e43b5141b14fc7cddf
# good: [e2e6f352889a6829b537f1b64ee3182bd27d31a1] Fix spiff and stdin_uart_interrupt overiding the same read function (#249)
git bisect good e2e6f352889a6829b537f1b64ee3182bd27d31a1
# skip: [c90362621a91c5b470c2f04725ec3bb4b3ffb1e9] Merge pull request #370 from UncleRus/extras/timeout_fix
git bisect skip c90362621a91c5b470c2f04725ec3bb4b3ffb1e9
# good: [0a1fd09459d5e95d5a3a73e4b243a02f9534cda7] lwip: revise ooseq handling.
git bisect good 0a1fd09459d5e95d5a3a73e4b243a02f9534cda7
# good: [b58cd12f8512c277f19859f675d10e1790efdbcb] Minor fixes in dhcpserver.c
git bisect good b58cd12f8512c277f19859f675d10e1790efdbcb
# good: [87a3503f931021f28d4dbe083f5dec08c626354b] I2C code formatting, README fix (#462)
git bisect good 87a3503f931021f28d4dbe083f5dec08c626354b
# bad: [5fa48d0298713f0124a7d738866d2eedcabcef2a] Code formatted, minor fixes (#466)
git bisect bad 5fa48d0298713f0124a7d738866d2eedcabcef2a
# bad: [8a474d749d69d21cdd09c8e0b3481744c01e3968] wificfg: correct a few initialization issues (#464)
git bisect bad 8a474d749d69d21cdd09c8e0b3481744c01e3968
# bad: [ebdd2f983bec275058addf7fed46bbbbd0c4dc50] lwip: define LWIP_POSIX_SOCKETS_IO_NAMES as zero (#457)
git bisect bad ebdd2f983bec275058addf7fed46bbbbd0c4dc50
# first bad commit: [ebdd2f983bec275058addf7fed46bbbbd0c4dc50] lwip: define LWIP_POSIX_SOCKETS_IO_NAMES as zero (#457)
```

funnydog, 2017-11-22:
Hi,

I'd like to propose the following patch as a fix (I can do a PR if needed):

```
diff --git a/extras/spiffs/esp_spiffs.c b/extras/spiffs/esp_spiffs.c
index 93ea24a..a541595 100644
--- a/extras/spiffs/esp_spiffs.c
+++ b/extras/spiffs/esp_spiffs.c
@@ -90,7 +90,7 @@ void esp_spiffs_init(uint32_t addr, uint32_t size)
     config.hal_write_f = esp_spiffs_write;
     config.hal_erase_f = esp_spiffs_erase;
 
-    config.fh_ix_offset = 3;
+    config.fh_ix_offset = SPIFFS_FILEHDL_OFFSET;
 
 }
 
```

ourairquality, 2017-11-22:
The fix looks good. Sorry that was missed in the patch.

funnydog, 2017-11-22:
Fine, would you prefer a proper PR with the fix?

ourairquality, 2017-11-22:
Yes a PR will be needed.

closed 2018-06-30 by kanflo