SuperHouse/esp-open-rtos #494: libc bug with locale.h

Zaltora, 2017-11-25:
I found a bug, i post here but not sure if it is related to esp-open-rtos.
I was trying `#include <locale.h>`
I want to change "." into "," when using float and printf, sprintf, ...
In my main, i do this : 
```
struct lconv* locale = localeconv();
locale->decimal_point = "," ;  //Printf float with a coma
```
After this, printf and sprintf work well, using coma instead of dot. but the function sscanf (maybe scanf  ? ) didn't work with float. Example  :
before (with dot config):  
=> sscanf( "123**.**123", "%f", &value)  =>  value = 123.123
after (with coma config) :
=> sscanf( "123**,**123", "%f", &value)  =>  value = 123.0
=> sscanf( "123**.**123", "%f", &value)  =>  value = 123.0

ourairquality, 2017-11-26:
This limitation is due to the default newlib being built to use the 'nano' io functions, see `stdio/nano-vfscanf_float.c` which hard codes the decimal point character. So if this is important you will need to recompile newlib configured not to use the nano variants, and then you might find that it does not link, overflows the iram, so might have the remove printf from iram too.

Zaltora, 2017-11-26:
If it is hard coded, why use dot with sscanf don't work ? ( this function call other things that need coma ? )
My solutions:
- hardcoded "coma"  line 183 and recompile ( but cant use dot )
- Switch between coma and dot when needed ( Warning: Need task thread safe with semaphore )
- Use nano-variants and remove printf from IRAM ( Not a problem, i dont use printf ). Where i can found the nano-variants settings ? ( Maybe try after finish my project, need code a lot of things )
 
Thank for your help!