SuperHouse/esp-open-rtos #496: write(2) to a socket freezes

imihajlow, 2017-11-26:
Hi all,

I'm not filing a bug here (I guess, you are not the addressee), but I would like to ask if anyone faced this problem and has a workaround.

I have a software which uses HTTP and HTTPS (via BearSSL). And sometimes writing to a socket freezes forever. I've figured out that this happens more often with large input buffers. I've implemented a workaround to feed `write()` with smaller pieces of data, and it worked until recently I've added a new task.

Does anyone here understands what is going on, how sockets are internally implemented and if a stable workaround exists? Maybe there's a way to check for free "socket buffer memory"?

ourairquality, 2017-11-26:
Socket descriptor 2 is not a valid lwip socket number but a write should have returned -1. The `write()` operation dispatches to `lwip_write()` in `core/newlib_syscalls.c` so it looks like an issue in lwip so look there. Perhaps a timeout needs to be defined or error handling improved etc. Lwip has a lot of debug output that can be enabled.

gpreviato, 2018-01-08:
I experienced a similar problem, but the reason was related to the heap becoming too small and unable to allocate some memory. Malloc failed silently somewhere, and the socket stuck...

imihajlow, 2018-01-08:
@gpreviato that's exactly what I've figured out. My workaround is to reduce heap usage.