SuperHouse/esp-open-rtos #498: HTTP / HTTPS OTA support

donny681, 2017-11-28:
Do esp-open-rtos support HTTP / HTTPS OTA support?I can't use espressif-ftop demo because the bootloader is different.

nochkin, 2017-11-28:
The extras/libesphttpd module supports OTA over HTTP.
There is an example in examples/esphttpd. You can build the OTA FW using "make ota" and flash it using "make webflash". You will need to set ESP_IP to your IP.

AndreaGreco, 2017-12-21:
I have recently write HTTP ota client, then reverse approach.
Esp made request, to HTTP server.
My version could also use, SHA256 for verify binary.
If you prefer this approach I can prepare an pull request.

donny681, 2018-01-06:
@AndreaGreco Greate,do you pull request?

AndreaGreco, 2018-01-06:
Not for now

AndreaGreco, 2018-01-22:
Done PR https://github.com/SuperHouse/esp-open-rtos/pull/553

AndreaGreco, 2018-07-03:
No i test it and it work with test file.
You could try to:
Enable debug in: extras/http_client_ota, 
Add "#define DEBUG" in `http_client_ota.c`
Then, add some debug.

From your log i can see that watch dog has reset it, after upgrade.

Look like that upgrade process work fine.

Are you sure that your initialization, is not too long?
In this system is necessary that idle task ping wdog.

Differently, you could send me your source, and i could try with it.

01matrix01, 2019-02-21:
Hi @AndreaGreco, there is a way to specify  "#define SERVER "192.168.1.30"" at run time? For example read data from sysparam?

Thank you

AndreaGreco, 2019-02-21:
Take look at example:
```examples/http_client_ota/http_get.c```

It build a static struct:
```
static ota_info info = {
    .server      = SERVER,
    .port        = PORT,
    .binary_path = BINARY_PATH,
    .sha256_path = SHA256_PATH,
};
```
You can compile your struct at runtime with strcpy.
So for example in your ```user_init``` create a function that compile ```ota_info struct``` by reading param from flash configuration area.

kiralikbeyin, 2019-08-04:
@AndreaGreco 

After 1 year later still cant start after upgrade process.

Tried with blink.bin and my.bin they throw same prints:

````
[OTA]	HTTP client task starting
[OTA]	Image will be saved in OTA slot 1
Error:Ota has completed upgrade process, all ready for system software reset
Delay 1
Delay 2
Delay 3
Reset
rm match
del if0
usl
sul 0 0

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x3ffe8000, len 744, room 16
tail 8
chksum 0xf0
load 0x40100000, len 2292, room 0

0x40100000: _stext at ??:?

tail 4
chksum 0xca
csum 0xca

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 1.

ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x3ffe8000, len 744, room 16
tail 8
chksum 0xf0
load 0x40100000, len 2292, room 0

0x40100000: _stext at ??:?

tail 4
chksum 0xca
csum 0xca

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 1.

````

AndreaGreco, 2019-08-06:
It same comments of a yeas ago.
No more debug, no more log noting I consider it work.
With this behavior sure not help other developer to support you.

I try it with nodemcu, and with this hardware still works.
Try check how your bootloader is configured or something like that.

Looks like that OTA has correctly write down firmware.
You can see this during last operation, of firmware before fist reset.
When bootloader bootstrap esp seems load from boot  area 16 this is strange is not correctly load from 2 or 1...., 16 is 0xFF for flash a uninizialized area.
Maybe a case but is strange.

Bootloader seems configured for big flash, if I remember correctly is impossible use OTA in big flash but still passed more the one year and I cannot remember about this details, maybe you can do your homework and check all this think before.
Read doc and Richard Burton bootloader a bit.
Try to understand how does work tftp OTA, http client arrive from tftp.

Can be good idea also try with nodemcu.
And report if it can work or not.
After this attack a complete log from OTA straup until bootloop

kiralikbeyin, 2019-08-07:
@AndreaGreco I tried **aboot** lib which flashed blink.bin in test directory 
https://github.com/SuperHouse/esp-open-rtos/issues/668

Prebuilt rboot or making rboot doesnt work but **aboot** only flashes this test blink.bin , stili cant flash other examples in esp-open-rtos..

I tried https://github.com/HomeACcessoryKid/life-cycle-manager/issues/11 which flashes only 560k 

I tried esphttpd example to upload a bin file but it is only for esp32..
https://github.com/SuperHouse/esp-open-rtos/issues/727

Is there any way to upload bin file by using a http server, select file and flash it?

kiralikbeyin, 2019-08-07:
@AndreaGreco Please can you send your rboot.bin file?