SuperHouse/esp-open-rtos #499: Malloc fails to allocate last 8K of RAM

maximkulkin, 2017-11-30:
I'm trying to do memory intensive operations on my ESP8266 (cryptography) and often run into out of memory issues so I have to be very careful. Profiling showed that cryptography requires ~22K of RAM. At start, I see that `xPortGetFreeHeapSize()` reports ~32K but sometimes operation still fails. It seems like if next allocation needs to allocate memory so that free heap size will go less than 8K, it fails.

Is there an explanation for this and/or ways to workaround this limitation and get access to last 8K. That would make code MUCH cleaner (and life easier).

ourairquality, 2017-11-30:
Probably fragmentation, which you could confirm by trying to allocate smaller chunks and seeing if that works. There is a block of unused memory that you might be able to use see `open_esplibs/libpp/wdec.c`.

maximkulkin, 2017-12-03:
I guess you're right.

closed 2017-12-03 by maximkulkin