SuperHouse/esp-open-rtos #506: SOFTAP_MODE auto-disconnect

Zaltora, 2017-12-07:
I have a problem since a while with the wifi.
When a client connects to the esp8266, after a certain time (approx 30 min (i will do more test about it)), the client is disconnected.
Did I miss an Lwip function to reset a watchdog or a timer?   It is a bug of lwip  ?  i got the same problem before the lwip update.

ourairquality, 2017-12-07:
Does it automatically reconnect? Is the issue that the connection gives up and stops re-connecting, or is this just a matter of a station leaving and re-joining after about 30 min?

Zaltora, 2017-12-07:
i think the client reconnect automatically to the wifi. ( when you enable ets_printf, you can see it ). But client need open a new socket connection.
I use putty to retrieve data over time ( for sensor analyze). When wifi disconect/reconnect. Putty stop and i need open a new socket connection . Problem i am not behind the screen.
I am not sur if lwip detect that the first socket was disconnect. I manage socket like the example
` esp-open-rtos/examples/tcp_non_blocking/tcp_non_blocking.c`
and sometime, i dont see events on socket when wifi was disable/enable. Socket connection become a ghost ^^

Zaltora, 2017-12-07:
My first problem that the wifi  disconnect/reconnect . It is normal ?
If i cant avoid this problem. What is the best way to detect it to close socket properly ?  
After a disconnect/reconnect. the client can talk again with old socket connection or need create a new one?

ourairquality, 2017-12-07:
A design that depends on a wifi connection staying up seems doomed, just not robust. Perhaps explore a design the caches the data on the device and allows it to be pulled or pushed in short exchanges.

Zaltora, 2017-12-07:
the wifi connection is not reliable even on a local network?  To add more detail, the wifi client is less than a meter. esp8266 got dhcp server + 2 listen port (one for command and one for real time data). Only one client is connected.
 I know that a wireless transmission cant be interrupt but with esp8266, the link will break. few hours maximum before link break.
i asking myself if sdk 0.9 or lwip got bug with a timeout. The wifi link from esp8266 feel instable when the link is establishes. the connection attempt too.
When a device try to connect to esp8266 , It can be fast or very long. I dont no why. Sometimes, esp8266 debug tell "max connections" but just one device want to connect. To avoid this, when i see that connection is too long to setup, i just do disconnect/reconnect  instantly(windows and linux) and the connection is ok.

i will do some rework to manage some short disconnect and close the old socket when a new from same ip is created. 
My app want to got 3 way to retreive data : UART (live)  / WIFI (live) and FILE (can be read in uart and wifi).

ourairquality, 2017-12-07:
It is probably riddled with bugs, sorry. There is not much we can do about them all, just work around the edges. We have made some good progress with the higher level layers, and there is more we can do. The softap mode appears to have more issues than the station mode. This might mean working with a somewhat flaky wifi layer, but if you design for robustness then it will probably help in other ways too, for example you might have a path to better power management, and be robust to working over longer distances in more congested environments etc.

Zaltora, 2017-12-08:
Damn , well i will try to do with it . Thank for your help.
Why espressive changed the code license. Why closed his souces. it hampers development :(
The official RTOS Sdk dont feel better too. Old freertos version, low frequency update.

But maybe wifi binary layer are fixes. We can't add theses binary in official project for license reason.
 It is hard for someone to add it in current project ? like take lib, remove some object and take few funtion. For example for wifi, take the wifi force api "fpm api"  or sleep system function  ?